carla-simulator/carla #976: make PythonAPI error win10

nicksunyang, 2018-11-19:


nicksunyang, 2018-11-19:
dependencies/include\carla/image/ImageIOConfig.h(81): error C2338: No image format supported, please compile with at least one of LIBCARLA_IMAGE_WITH_PNG_SUPPORT, LIBCARLA_IMAGE_WITH_JPEG_SUPPORT, or LIBCARLA_IMAGE_WITH_TIFF_SUPPORT
dependencies/include\carla/image/ImageIOConfig.h(299): error C2338: Default IO needs to be supported.
dependencies/include\carla/image/ImageIOConfig.h(320): note: 参见对正在编译的 类 模板 实例化 "carla::image::io::detail::io_any<carla::image::io::detail::io_jpeg>" 的引用
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Professional\\VC\\Tools\\MSVC\\14.15.26726\\bin\\HostX86\\x64\\cl.exe' failed with exit status 2

iFuSiiOnzZ, 2018-11-20:
Hi,

Sorry for the delay. Currently building PythonAPI in Windows is a bit painful as it has dependencies on external libraries as [LibPNG](http://gnuwin32.sourceforge.net/packages/libpng.htm) which depends on [ZLIB](https://zlib.net/). So, for now, I made some quick scripts to build the dependencies and manually copy the binaries to the PythonAPI dependencies folder.

Files needed to update: [PythonAPI\setup.py](https://pastebin.com/uzqmb7Mb)
* Includes added: png, zlib
* Libs added: boost filesystem, libpng and zlib
* Compilation defines: LIBCARLA_IMAGE_WITH_PNG_SUPPORT=true

Next step is to build [libpng](https://pastebin.com/LgsnYQkQ) {batch file} and [zlib](https://pastebin.com/Qe2hTpEU) {powershell file}.
* Put both files into a folder
* Execute the powershell script (will build zlib and execute the script in charge of building libpng)
* If everything when OK there will be a folde called `bin` where you will find the includes for libpng and zlib and the lib folder for the binaries.
* copy `bin\png` and `bin\zlib` to `PythonAPI\dependencies\include` (keep folder structure)
* copyt `bin\libs\*` to `PythonAPI\dependencies\lib`

If the folder `PythonAPI\dependencies\` doesn't exist execute first `make LibCarla`. Now that we have everything executing `make PythonAPI` it should build with no problem the egg file.

salcf, 2018-11-23:
Hi,

I have managed to build zlib using the powershell script provided.
However, when trying to build libpng using the batch script I am receiving the following error:
**\BuildWin64\Intermediate\png.obj : fatal error LNK1112: module machine type 'x86' conflicts with target machine type 'x64'**
Do you know how this problem can be solved?

Thanks in advance!

iFuSiiOnzZ, 2018-11-23:
PowerShell script should also execute batch file. Also important, load the command line tools for x64 and execute the powershell script from command line as there is no easy way to do it from powershell.

salcf, 2018-11-26:
I solved that error, but now, when I execute make PythonAPI I get the following:
LINK : fatal error LNK1181: cannot open input file 'D:\FixedCarla\carla\PythonAPI\dependencies\lib\libboost_python37-vc141-mt-x64-1_67.lib'
I looked in my lib directory and found that I have libboost_python36-vc141-mt-x64-1_67.lib
Have I done something wrong when building CARLA for the first time or something?

Thanks in advance!

iFuSiiOnzZ, 2018-11-26:
It sounds like you have both versions of python installed, is it right? And while boost is builded is using python3.6 and 3.7 for pythonapi. Take a look at the windows environment variable PATH to see if something strange there (remember that you have one per user and the system one).

delrocco, 2018-11-29:
```-[BuildLibCarla]: LibCarla server has been successfully installed in "C:/Users/jdelrocc/Documents/carla/Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies"!
-[BuildLibCarla]: LibCarla client has been successfully installed in "C:/Users/jdelrocc/Documents/carla/PythonAPI\dependencies"!
-[BuildPythonAPI]: [Batch params]: --py3
INFO: Could not find files for the given pattern(s).

-[BuildPythonAPI]: [ERROR] An error ocurred while executing the py.
-[BuildPythonAPI]: [ERROR] Possible causes:
-[BuildPythonAPI]: [ERROR]  - Make sure "py" is installed.
-[BuildPythonAPI]: [ERROR]  - Make sure it is available on your Windows "py".
make: *** [PythonAPI] Error 1```

If I pip list, "py" module is installed.

M-Hammod, 2018-11-30:
@Andrei Jianu, 

thank you for providing this, I had to do the following for me to work : 

1- add  **Set-ExecutionPolicy -Scope Process -ExecutionPolicy Bypass** at the beginning of zlib.ps1
2- install install v140 build tools as explained in the following link https://developercommunity.visualstudio.com/content/problem/48806/cant-find-v140-in-visual-studio-2017.html
3- as you mentioned I ran the command powershell ./zlib.ps1 from inside the x64 native tools command prompt for VS2017

Thanks again

iFuSiiOnzZ, 2018-11-30:
Thank you @M-Hammod, I updated my post making reference to yours in case of any problem related to powershell permissions or missing build tools.

sriharshakunda, 2018-12-28:
I am having the same issue as @delrocco. Can someone help me what am I doing wrong?
I have python 3.5.5 installed.

sriharshakunda, 2018-12-29:
I am having an issue while building PythonAPI.
Below is my error log.

c:\carla\pythonapi\source\libcarla\Blueprint.cpp(155): error C2672: 'boost::python::class_<carla::client::BlueprintLibrary,boost::noncopyable,boost::shared_ptrcarla::client::BlueprintLibrary,boost::python::detail::not_specified>::def': no matching overloaded function found c:\carla\pythonapi\source\libcarla\Blueprint.cpp(155): error C2780: 'boost::python::class_<carla::client::BlueprintLibrary,boost::noncopyable,boost::shared_ptrcarla::client::BlueprintLibrary,boost::python::detail::not_specified> &boost::python::class_<carla::client::BlueprintLibrary,boost::noncopyable,boost::shared_ptrcarla::client::BlueprintLibrary,boost::python::detail::not_specified>::def(const char *,Fn,const A1 &,const A2 &,const A3 &)': expects 5 arguments - 1 provided dependencies/include\boost/python/class.hpp(261): note: see declaration of 'boost::python::class_<carla::client::BlueprintLibrary,boost::noncopyable,boost::shared_ptrcarla::client::BlueprintLibrary,boost::python::detail::not_specified>::def' c:\carla\pythonapi\source\libcarla\Blueprint.cpp(155): error C2780: 'boost::python::class_<carla::client::BlueprintLibrary,boost::noncopyable,boost::shared_ptrcarla::client::BlueprintLibrary,boost::python::detail::not_specified> &boost::python::class_<carla::client::BlueprintLibrary,boost::noncopyable,boost::shared_ptrcarla::client::BlueprintLibrary,boost::python::detail::not_specified>::def(const char *,Fn,const A1 &,const A2 &)': expects 4 arguments - 1 provided dependencies/include\boost/python/class.hpp(245): note: see declaration of 'boost::python::class_<carla::client::BlueprintLibrary,boost::noncopyable,boost::shared_ptrcarla::client::BlueprintLibrary,boost::python::detail::not_specified>::def' c:\carla\pythonapi\source\libcarla\Blueprint.cpp(155): error C2780: 'boost::python::class_<carla::client::BlueprintLibrary,boost::noncopyable,boost::shared_ptrcarla::client::BlueprintLibrary,boost::python::detail::not_specified> &boost::python::class_<carla::client::BlueprintLibrary,boost::noncopyable,boost::shared_ptrcarla::client::BlueprintLibrary,boost::python::detail::not_specified>::def(const char *,A1,const A2 &)': expects 3 arguments - 1 provided dependencies/include\boost/python/class.hpp(238): note: see declaration of 'boost::python::class_<carla::client::BlueprintLibrary,boost::noncopyable,boost::shared_ptrcarla::client::BlueprintLibrary,boost::python::detail::not_specified>::def' c:\carla\pythonapi\source\libcarla\Blueprint.cpp(155): error C2780: 'boost::python::class_<carla::client::BlueprintLibrary,boost::noncopyable,boost::shared_ptrcarla::client::BlueprintLibrary,boost::python::detail::not_specified> &boost::python::class_<carla::client::BlueprintLibrary,boost::noncopyable,boost::shared_ptrcarla::client::BlueprintLibrary,boost::python::detail::not_specified>::def(const char *,F)': expects 2 arguments - 1 provided dependencies/include\boost/python/class.hpp(229): note: see declaration of 'boost::python::class_<carla::client::BlueprintLibrary,boost::noncopyable,boost::shared_ptrcarla::client::BlueprintLibrary,boost::python::detail::not_specified>::def' c:\carla\pythonapi\source\libcarla\Blueprint.cpp(155): error C2784: 'boost::python::class_<carla::client::BlueprintLibrary,boost::noncopyable,boost::shared_ptrcarla::client::BlueprintLibrary,boost::python::detail::not_specified> &boost::python::class_<carla::client::BlueprintLibrary,boost::noncopyable,boost::shared_ptrcarla::client::BlueprintLibrary,boost::python::detail::not_specified>::def(const boost::python::def_visitor &)': could not deduce template argument for 'const boost::python::def_visitor &' from 'const char [12]' dependencies/include\boost/python/class.hpp(219): note: see declaration of 'boost::python::class_<carla::client::BlueprintLibrary,boost::noncopyable,boost::shared_ptrcarla::client::BlueprintLibrary,boost::python::detail::not_specified>::def' c:\carla\pythonapi\source\libcarla\Blueprint.cpp(156): error C2228: left of '.def' must have class/struct/union c:\carla\pythonapi\source\libcarla\Blueprint.cpp(156): fatal error C1003: error count exceeds 100; stopping compilation error: command 'C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\x86_amd64\cl.exe' failed with exit status 2 -[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:/carla/PythonAPI\dist"!

However, when I look for "dist" folder in PythonAPI folder, there is none.

Can someone point me what am I doing wrong?

I did fresh installations twice and couldn't find whats wrong.

sriharshakunda, 2019-01-02:
@sagnibak I had the same problem. The only successful build I had so far was 0.9.0 (https://github.com/carla-simulator/carla/tree/fix_vs2017_build_solution). I believe 0.9.X, PythonAPI is still work in progress on windows platform.

didibaba, 2019-01-18:
@sriharshakunda @sagnibak I had exactly the same problems as you two.  Someone mentioned branch fix_vs2017_build_solution does not support server. Any progress made so far?

zoeeer, 2019-01-25:
Hi @iFuSiiOnzZ ,

Thanks a lot for your scripts.
Though I failed building zlibstat.lib using the script directly (got strange internal compiler errors), I opened the VS project it generated and built the lib from there.

Still working on `make PythonAPI`. I'm getting similar error as @andreimihalcea97's [comment](https://github.com/carla-simulator/carla/issues/976#issuecomment-441682449).

gaurish123, 2019-01-29:
> I have Python 3.5 and 2.7 installed.
> `make launch` seems to build without error now.
> `make LibCarla` builds.
> `make PythonAPI` is not building...
> The [ps1](https://pastebin.com/LgsnYQkQ) and [bat](https://pastebin.com/Qe2hTpEU) files above build successfully, and I copied over the includes and libs.
> But I still get this:
> 
> ```
> -[BuildLibCarla]: LibCarla server has been successfully installed in "C:/Users/jdelrocc/Documents/carla/Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies"!
> -[BuildLibCarla]: LibCarla client has been successfully installed in "C:/Users/jdelrocc/Documents/carla/PythonAPI\dependencies"!
> -[BuildPythonAPI]: [Batch params]: --py3
> INFO: Could not find files for the given pattern(s).
> 
> -[BuildPythonAPI]: [ERROR] An error ocurred while executing the py.
> -[BuildPythonAPI]: [ERROR] Possible causes:
> -[BuildPythonAPI]: [ERROR]  - Make sure "py" is installed.
> -[BuildPythonAPI]: [ERROR]  - Make sure it is available on your Windows "py".
> make: *** [PythonAPI] Error 1
> ```
> If I pip list, "py" module is installed.
> 
> Is my Python configuration not setup right for Carla build system? Am I missing necessary environment variables or something?

Hi, I am trying to build pythonapi for carla ,, I download the two file mentioned above
https://pastebin.com/LgsnYQkQ
https://pastebin.com/Qe2hTpEU

but when I am trying to excecute it, getting this error
C:\MinGW\fix2>& $BuildPNGBAT

C:\MinGW\fix2>.\Qe2hTpEU.bat

C:\MinGW\fix2>New-Item -ItemType Directory -Path ".\bin"
'New-Item' is not recognized as an internal or external command,
operable program or batch file.

C:\MinGW\fix2>New-Item -ItemType Directory -Path ".\bin\png"
'New-Item' is not recognized as an internal or external command,
operable program or batch file.

  Do I need to run the file at some specific location. please help me here.

sumeetambastha, 2019-01-29:
@gaurish123, I am able to generate the binary files. 
These are the steps you should follow:
1. Download the scripts and rename to png.bat and zlib.ps1 (text file to zlib.ps1, make sure you get rid of extension .txt) respectively.
2. If not installed, install v140 build tools as explained step by step in the following link: https://developercommunity.visualstudio.com/content/problem/48806/cant-find-v140-in-visual-studio-2017.html (Thanks to @M-Hammod , this is the most important step).
3. Once installed, open "x64 Native Command Tool for Visual Studio 2017".
4. cd to the directory where you have these files. Make sure you are connected to internet.
5. Run this command (without quotes):   "powershell -ExecutionPolicy ByPass ./zlib.ps1"
6. This script is supposed to run "png.bat", if that fails because of some reason, you can run that by double clicking it.
7. You can find the files in all the 3 folders.
8. Cheers!

eds89, 2019-02-03:
> Hi,
> 
> Sorry for the delay. Currently building PythonAPI in Windows is a bit painful as it has dependencies on external libraries as [LibPNG](http://gnuwin32.sourceforge.net/packages/libpng.htm) which depends on [ZLIB](https://zlib.net/). So, for now, I made some quick scripts to build the dependencies and manually copy the binaries to the PythonAPI dependencies folder.
> 
> Files needed to update: [PythonAPI\setup.py](https://pastebin.com/uzqmb7Mb)
> 
> * Includes added: png, zlib
> * Libs added: boost filesystem, libpng and zlib
> * Compilation defines: LIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
> 
> Next step is to build [libpng](https://pastebin.com/LgsnYQkQ) {batch file} and [zlib](https://pastebin.com/Qe2hTpEU) {powershell file}.
> 
> * Put both files into a folder
> * Execute the powershell script (will build zlib and execute the script in charge of building libpng)
> * If everything when OK there will be a folde called `bin` where you will find the includes for libpng and zlib and the lib folder for the binaries.
> * copy `bin\png` and `bin\zlib` to `PythonAPI\dependencies\include` (keep folder structure)
> * copyt `bin\libs\*` to `PythonAPI\dependencies\lib`
> 
> If the folder `PythonAPI\dependencies\` doesn't exist execute first `make LibCarla`. Now that we have everything executing `make PythonAPI` it should build with no problem the egg file.
> 
> IMPORTANT!!!
> 
> * Run the powershell script from the command line (x64 build tools), it will execute the bat file also
> * If you have any problem with pawershell permissions or v140 build tools missing take a look at the @M-Hammod  [comment](https://github.com/carla-simulator/carla/issues/976#issuecomment-443082525)

What powershell module does the 'MsBuild' command belong from?

shriram17, 2019-02-05:
When I attempt to make PythonAPI on following the steps mentioned, using the Carla 0.9.2 nsubiron/windows_build_wip branch and Python 3.6
LINK : fatal error LNK1181: cannot open input file -'D:\Simulators\Carla_9_2\carla\PythonAPI\dependencies\lib\libboost_filesystem-vc141-mt-x64-1_67.lib',
file 'libboost_filesystem-vc141-mt-x64-1_69.lib' exists in the dependencies folder. 
Any suggestion on how to solve this ??

zoeeer, 2019-02-15:
> Still working on `make PythonAPI`. I'm getting similar error as @andreimihalcea97's [comment](https://github.com/carla-simulator/carla/issues/976#issuecomment-441682449).

I think Python 3.5 on Windows has some issues. I replaced it with Python 3.6 and now I have successfully built PythonAPI.

backer777, 2019-02-27:
> * Compilation defines: LIBCARLA_IMAGE_WITH_PNG_SUPPORT=true

My problem now is that I do not know in which file to add this option `LIBCARLA_IMAGE_WITH_PNG_SUPPORT=true` . Thank you

closed 2019-02-27 by nsubiron

osama-usuf, 2019-06-17:
@iFuSiiOnzZ Your scripts on PasteBin seem to have been removed. Can you please provide a mirror?

JJW152, 2019-10-08:
This seems to be a legacy issue but I wanted to reiterate the solution to @delrocco 's problem.

I have Python installed in Windows through Anaconda, and I put Python in my PATH, which is pretty common. The python launcher doesn't come in the Anaconda distribution, and thus "py" is not a recognized command.

This is solved by replacing "py -3" with "python" in the PythonAPI .bat file.

adi-geek2, 2019-10-14:
> @iFuSiiOnzZ Your scripts on PasteBin seem to have been removed. Can you please provide a mirror?

+1

urishab, 2019-12-04:
This might be obvious to many but "py" is actually a reference to "py launcher". 
This is a utility bundled with Python on Windows but not installed by default.
I ran the Python installation and it allowed me to "modify" my installation and install "py launcher".
I'll submit a PR with a better error message

tw0226, 2021-06-22:
@delrocco @JJW152 I also had the error.

```
-[BuildLibCarla]: LibCarla server has been successfully installed in "C:/Users/jdelrocc/Documents/carla/Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies"!
-[BuildLibCarla]: LibCarla client has been successfully installed in "C:/Users/jdelrocc/Documents/carla/PythonAPI\dependencies"!
-[BuildPythonAPI]: [Batch params]: --py3
INFO: Could not find files for the given pattern(s).

-[BuildPythonAPI]: [ERROR] An error ocurred while executing the py.
-[BuildPythonAPI]: [ERROR] Possible causes:
-[BuildPythonAPI]: [ERROR]  - Make sure "py" is installed.
-[BuildPythonAPI]: [ERROR]  - Make sure it is available on your Windows "py".
make: *** [PythonAPI] Error 1
```

In BuildPythonAPI, change explicitly py -> python3 

```
where py 1 > nul     >>>> where python3 1>nul
```

In my case, I solved it! :)

Liuzy0908, 2021-10-01:
> @delrocco @JJW152 I also had the error.
>
> ```
> -[BuildLibCarla]: LibCarla server has been successfully installed in "C:/Users/jdelrocc/Documents/carla/Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies"! [...]
Hi, I'm compiling Cara 0.9.12 and I didn't find the BuildPythonAPI you said. Can you tell me where it is? Many thanks.

molua, 2023-07-13:
> I have Python 3.5 and 2.7 installed. `make launch` seems to build without error now. `make LibCarla` builds. `make PythonAPI` is not building... The [ps1](https://pastebin.com/LgsnYQkQ) and [...]

have you solve the problem？

2019211753, 2023-10-28:
> @delrocco @JJW152 I also had the error.
>
> ```
> -[BuildLibCarla]: LibCarla server has been successfully installed in "C:/Users/jdelrocc/Documents/carla/Unreal\CarlaUE4\Plugins\Carla\CarlaDependencies"! [...]

it solved for me in 2023, thanks😄

nadrainam, 2023-12-15:
where can we find this  PythonAPI .bat file.in which we can make changes. can someone please specify the folder name

blacksilvergeek, 2025-03-25:
@sagnibak 

I believe I have successfully resolved the issue. Initially, after modifying it to use Python, the program runs, but nothing was produced.

In my case, I found two bugs that needed to be addressed:

1. Xerces Error
   The program reported an error related to Xerces, but it did not exit. You can download the necessary file from the link below and manually place it into the build directory:  
   [Download Xerces-C 3.2.3](https://archive.apache.org/dist/xerces/c/3/sources/xerces-c-3.2.3.tar.gz)

2. Missing Files
   Although no files were generated initially, you can navigate to the `PythonAPI/carla` folder and run one of the following commands:  
   - `python setup.py bdist_wheel`  
   - `python setup.py bdist_egg`  
   
   This will create the whl or egg file in the `dist` folder.

PR #1241 (2019-02-27): Basic build system for Windows
#### Description
A basic build system for Windows.

**This PR provides**
- Support for Python3 (`make PythonAPI`)
- Automated download and build of `zlib` and `libpng` needed in Win
- Now Boost is downloaded from `https` and not from git

**This PR does NOT provides**
- Support for Python2 (not tested really)
- Support for Google Test

Fixes #1000 
Fixes #1155
Fixes #976

#### Full test of functionalities
1. Open a `Visual Studio x64 Native Tools Command Prompt` and navigate to the project folder
1. Remove `carla/Build` folder (if exists)
1. Run `make clean launc [... cut]