carla-simulator/carla #977: Bone joint Coordinate

Nozomeister, 2018-11-20:
Hi Carla team,
From what I know, we can extract the pedestrian coordinate through the coordinate of their static mesh, but can we further extract the coordinate or data of their bone joint or specific arms/legs?
Thank you.

TheNihilisticRobot, 2018-11-20:
Hi @Nozomeister 

That's not yet possible,  we plan  to implement it in some future, though that future isn't near.

Nozomeister, 2018-11-22:
Thank you for the response. When exploring around I tried to modify a pedestrian mesh blueprint and I manage to print out some part of the information on the screen/UE4 log. Do you have any suggestion so that I can print out them on a different text file and print in on every single frame like the coordination from client_example.py code?
Thank you
[image]

closed 2019-01-28 by stale[bot]

jagdishbhanushali, 2019-03-07:
Hi @Nozomeiste, I am also working on same problem, getting bone joints of pedestrians. Can you please let me know how you managed to get joints in Unreal editor as well as in PythonAPI.

Thanks.

YDSh510, 2019-05-20:
Hi @Nozomeister and @TheNihilisticRobot 

I also want to add some animation in the pedestrian. Can you please let me know how you managed to get joints and which blueprint file are you referring to here?

hvelesaca, 2021-05-17:
is there an update on this problem, I would like to get the bone joints coordinate from a walker