carla-simulator/carla #982: Packaging issue in Docker

Danmou, 2018-11-20:
I'm trying to create a Dockerfile to build Carla. I'm using the nvidia/opengl:1.0-glvnd-devel-ubuntu16.04 Docker image as a base and following exactly the steps in the documentation to build Carla. I got both Unreal Engine and Carla to build successfully, but when I try to generate the packaged version (using `Package.sh` or `make package` depending on the Carla version), I get the following errors:
```
  1001LogLinker: Warning: Can't find file '/carla/carla/Unreal/CarlaUE4/Content/Blueprints/Game/CarlaGameMode.uasset'
  LogUObjectGlobals: Warning: Failed to load '/carla/carla/Unreal/CarlaUE4/Content/Blueprints/Game/CarlaGameMode.uasset': Can't find file '/carla/carla/Unreal/CarlaUE4/Content/Blueprints/Game/CarlaGameMode.uasset'
  CookResults: Error: Error loading /carla/carla/Unreal/CarlaUE4/Content/Blueprints/Game/CarlaGameMode.uasset!
  LogCook: Error: Error loading /carla/carla/Unreal/CarlaUE4/Content/Blueprints/Game/CarlaGameMode.uasset!
  1001LogLinker: Warning: Can't find file '/carla/carla/Unreal/CarlaUE4/Content/Maps/Town01.umap'
  LogUObjectGlobals: Warning: Failed to load '/carla/carla/Unreal/CarlaUE4/Content/Maps/Town01.umap': Can't find file '/carla/carla/Unreal/CarlaUE4/Content/Maps/Town01.umap'
  CookResults: Error: Error loading /carla/carla/Unreal/CarlaUE4/Content/Maps/Town01.umap!
  LogCook: Error: Error loading /carla/carla/Unreal/CarlaUE4/Content/Maps/Town01.umap!
  1001LogLinker: Warning: Can't find file '/carla/carla/Unreal/CarlaUE4/Content/Maps/Town02.umap'
  LogUObjectGlobals: Warning: Failed to load '/carla/carla/Unreal/CarlaUE4/Content/Maps/Town02.umap': Can't find file '/carla/carla/Unreal/CarlaUE4/Content/Maps/Town02.umap'
  CookResults: Error: Error loading /carla/carla/Unreal/CarlaUE4/Content/Maps/Town02.umap!
  LogCook: Error: Error loading /carla/carla/Unreal/CarlaUE4/Content/Maps/Town02.umap!
```

The files it mentions do exist and are owned by the `carla` user (which is also the one running the package command). `/carla` is the home directory of this user. I have tried both Carla 0.8.4 and 0.9.0.

Any idea what is going wrong?

ajboloor, 2018-11-24:
You ideally want to first package your Carla project using Unreal, or ``make package`` as you mentioned.
TEST this (./CarlaUE4.sh) without using the docker to see if it works as expected.
Then in your packaged directory (typically under LinuxNoEditor/), put the docker file:
Dockerfile
```
# Make sure drivers are >= 390
# sudo docker run -p 2000-2002:2000-2002 --runtime=nvidia -e NVIDIA_VISIBLE_DEVICES=ID carla:latest ./CarlaUE4.sh 
# /Game/Maps/Town01 -benchmark -carla-server -fps=10 -world-port=2000 -windowed -ResX=100 -ResY=100 -carla-no-hud

FROM nvidia/opengl:1.0-glvnd-runtime-ubuntu16.04

RUN apt-get update; apt-get install -y libsdl2-2.0

RUN useradd -m carla
USER carla
ENV HOME /home/carla
COPY . /home/carla

USER root
RUN chown -R carla:carla /home/carla

RUN apt-get -y install sudo
RUN echo "carla:carla" | chpasswd && adduser carla sudo

USER carla
WORKDIR /home/carla

ENV SDL_VIDEODRIVER=offscreen

CMD /bin/bash CarlaUE4.sh -carla-server
```
Then run:
`sudo docker build -t carla-friendly-tag .`
This will build the docker image. To use it, run something like:
`sudo docker run -p 2000-2002:2000-2002 --runtime=nvidia -e NVIDIA_VISIBLE_DEVICES=0 carla-friendly-tag ./CarlaUE4.sh`
Source: https://carla.readthedocs.io/en/latest/carla_docker/

Danmou, 2018-11-24:
Thanks for the advice. However, we would like to automate the build process as part of our CI/CD pipeline, so it would be ideal if everything could be done in Docker in a repeatable fashion. The build of both UE and Carla succeeds, only the packaging step fails due to this strange error.

closed 2019-01-31 by stale[bot]

weijiepengx, 2022-06-08:
meet same issue in `carla-9.13`.

weijiepengx, 2022-06-10:
> meet same issue in `carla-9.13`.
Issue fixed. Seems a bug of UnrealEngine. `Content` of Carla is located at `${CARLA_ROOT}/Unreal/CarlaUE4/Content/Carla`. If  any of folders in `/` which name is same as folder in `${CARLA_ROOT}/Unreal/CarlaUE4/Content` , error occurs.
In my Docker container, there exists a folder named `Carla` in `/`, I cooked failed and got error messages as above.
Workaround for this is to remove or rename `/Carla`  folder.