carla-simulator/carla #985: Unable to run Python Example codes in CARLA 0.9.1

YashBansod, 2018-11-21:
I downloaded the Pre-Compiled version of CARLA 0.9.1. Extracted it in a directory.
Then I started CARLA in server mode: `./CarlaUE4.sh`.
As per the change log of the release - https://github.com/carla-simulator/carla/blob/master/CHANGELOG.md#carla-091
The example.py is now tutorial.py

So, in a new terminal I type: `python tutorial.py`.
I get the following error:

> Traceback (most recent call last):
>   File "tutorial.py", line 21, in <module>
>     import carla
>   File "PythonAPI/carla-0.9.1-py2.7-linux-x86_64.egg/carla/__init__.py", line 7, in <module>
>   File "PythonAPI/carla-0.9.1-py2.7-linux-x86_64.egg/carla/libcarla.py", line 7, in <module>
>   File "PythonAPI/carla-0.9.1-py2.7-linux-x86_64.egg/carla/libcarla.py", line 6, in __bootstrap__
> ImportError: libpng16.so.16: cannot open shared object file: No such file or directory

I tried running: `python manual_control.py`.
I get the following error:

> Traceback (most recent call last):
>   File "manual_control.py", line 63, in <module>
>     from PythonAPI import carla
> ImportError: No module named PythonAPI

I am sure I followed all the instructions in the Getting Started page of documentation to word.
https://carla.readthedocs.io/en/latest/getting_started/

Can someone help me with what I am doing wrong?

Marwa215, 2018-11-21:
I think you should to use python3 instead of python2 as Carla9.1 support python3 right now.

fcagroupj, 2018-11-21:
try this: https://github.com/fcagroupj/fcarsim

YashBansod, 2018-11-22:
To run the example code - **manual_control.py**, you also need to go inside PythonAPI directory and extract carla-0.9.1-py2.7-linux-x86_64.egg there. 
Then you need to copy the contents of the extracted folder back into PythonAPI directory.
The PythonAPI directory should look like this:
- PythonAPI
    - carla
    - EGG-INFO

And then I was able to run it by `python3 manual_control.py`.

YashBansod, 2018-11-22:
The `python manual_control.py` Still gives the error though:
Things I tried: 
- extracting the carla-0.9.1-py2.7-linux-x86_64.egg
- Adding the following to Python Path: `export PYTHONPATH=./:PythonAPI/`

But the error remains the same

danil-tolkachev, 2018-11-22:
@YashBansod Are you sure that you are using right version of manual_control.py? I've checked `PythonAPI/manual_control.py` (**not** `Deprecated/PythonClient/manual_control.py`) and there's no line `from PythonAPI import carla`.

YashBansod, 2018-11-22:
@danil-tolkachev I downloaded the Pre-Compiled version of CARLA 0.9.1. The **manual_control.py** in the **precompiled** version has that line. Its because in the Precompiled version, the **manual_control.py** is outside the PythonAPI directory.

danil-tolkachev, 2018-11-22:
@YashBansod It's strange. In the precompiled version (0.9.1 from https://github.com/carla-simulator/carla/releases) I don't see such lines.  Also, in the PythonAPI i have *.egg files, so it doesn't require __init_.py.

autonomobil, 2019-08-19:
I had a similar issue in a conda enviroment. This was my solution:
 
- locate .egg-files in `Carla/PythonAPI/carla/dist/`
- change/install python version in conda enviroment which fits closest to an .egg-file (for me carla-0.9.6-py3.5-linux-x86_64.egg -> 3.5)
- delete all other .egg-files (just for tidiness)
- extract your .egg-file, e.g. carla-0.9.6-py3.5-linux-x86_64
- create `setup.py` inside the folder (carla-0.9.6-py3.5-linux-x86_64) with the following content:

```
from distutils.core import setup
setup(name='carla',
      version='0.9.6', #doesn't matter I guess
      py_modules=['carla'],
      )
```
- then you can install the carla-python-package via pip: 
`pip install -e ~/Carla/PythonAPI/carla/dist/carla-0.9.6-py3.5-linux-x86_64`

That solved it for me. I don't even need the egg-path-appending-stuff, which is happening at the beginning of every python-script, just `import carla.`