carla-simulator/carla #986: How to run CARLA 0.9.1 in Game Mode

YashBansod, 2018-11-21:
I downloaded the Pre-Compiled version of CARLA 0.9.1. Extracted it in a directory.
Then I started CARLA in server mode: `./CarlaUE4.sh`.
As mentioned in the documentation: https://carla.readthedocs.io/en/latest/getting_started/
CARLA starts in the server mode. 

**I wanted to know if there was any way to run CARLA in the game mode.**

I tried `./CarlaUE4.sh -help` and a few more options, but am unable to get the help message of find any documentation on how to run CARLA in game mode.

fcagroupj, 2018-11-21:
After staring CARLA server, you'd run client application like manual_control.py

YashBansod, 2018-11-22:
@fcagroupj thanks for the reply. I was facing some issue with #985 so I was looking for a way to directly run CARLA in game mode. But now that its resolved your suggested way works.

So I am assuming that you **cannot** run CARLA directly in Game Mode. 
You will always have to run it in server mode and control it using a client - through manual_control.py.
Am I correct in understanding this?

marcgpuig, 2018-11-26:
> Am I correct in understanding this?

@YashBansod Yes you are. There is a unique Mode now in Carla :)

closed 2018-12-05 by YashBansod