carla-simulator/carla #990: No module named 'carla' after tutorial.py execution

erivera1802, 2018-11-23:
Hello Everyone,
I am triying to drive manually a car with the 0.9.0 version of carla, I have read that it isnt just running it in Unreal, but you have to execute the python files for it. So I tried exactly that, but  it doesnt find the carla python modules:
```

esteban@esteban-Hyrican-PC:~/carla/PythonAPI$ ./tutorial.py
Traceback (most recent call last):
  File "./tutorial.py", line 21, in <module>
    import carla
ModuleNotFoundError: No module named 'carla'
```
The same error appeared when I executed manual_control.py (I think that is what I wan now, to manually control a vehicle).

I have already followed the instructions of the readthedocs page, and can already see carla working over unreal, basically exactly like this:
https://github.com/carla-simulator/carla/issues/660
Does anyone know what am I missing?

If you are reporting an issue, please provide the CARLA version number you are
using and your Platform/OS.

-->

fabianoboril, 2018-11-28:
Hi,

if you want to use the Python clients you should run "make PythonAPI". This will build the carla modules (you can find them in PythonAPI/dists/*.egg). Then you either install them (e.g. with easy_install) or for testing extend your PYTHON_PATH to include these. Hope it works.

Bests

tanglrHello, 2018-11-29:
Hi, you can run "make package" at the root path of carla. If succeed, you will get a 'Dist' directory under this path. Enter it and you will find the .py files. Run the .py files at this place and it should be ok.

erivera1802, 2018-11-30:
Hi, @fabianoboril and @tanglrHello,
I tried the 'make PythonAPI' and 'make package' commands, but again i get the same error: carla/PythonAPI/dependencies/lib/libboost_python37.a: No such file or directory.
Until now, my workaround has been executing them with python2, but I though this 0.9.1 version was already compatible with python3.

fabianoboril, 2018-11-30:
Can you maybe post a compile log that we could look into? Thanks!

yyuuliang, 2018-12-06:
Maybe try the setup.py in the PythonAPI folder.

coreper, 2018-12-09:
> Hi, @fabianoboril and @tanglrHello,
> I tried the 'make PythonAPI' and 'make package' commands, but again i get the same error: carla/PythonAPI/dependencies/lib/libboost_python37.a: No such file or [...]

Same issue for me.

manishthani, 2019-01-08:
Hi everyone,
Having Anaconda installed for Python3 shows exactly the same error when trying to run `make PythonAPI` command. Solved the problem by uninstalling it following the steps described in this [link](https://stackoverflow.com/questions/22585235/python-anaconda-how-to-safely-uninstall) and executing the command again.
Hope this helps!

closed 2019-03-16 by stale[bot]

MaxHBor, 2019-04-19:
Hello. Got the same problem with compiled 0.9.4 version for Windows.