carla-simulator/carla #991: Index Error: bytes to write exceed buffer size, error on running manual_control.py with Carla 0.9.1

aroongta, 2018-11-26:
Hi, I am facing a strange error when trying to run the manual_control.py script with Carla 0.9.1.
It shows an index error: bytes to write exceed buffer size in the terminal. 
The pygame window is still running with server fps:15-20 fps and client fps: 62, though there is a lot of lag in the response observed on the pygame window on pressing any input keys. I am not sure if this error is doe to some problem in my PC or in the script. I am using the manual_control.py script in the PythonAPI folder. 
[image]

fabianoboril, 2018-11-26:
Hi,

we had the same issue a few days ago with 0.9.1. In our case the issue it turned out that during build Carla and the PythonAPI, Python2 and Python3 libraries were mixed. You can test if "python2 manual_control.py" or "python3 manual_control.py" resolves the issue. 

Bests

aroongta, 2018-11-26:
thanks @fabianoboril . 
It turned out I was using the older version of pygame, I updated that and using python 3, after that is working just fine now.

closed 2018-11-27 by marcgpuig

togaen, 2019-03-24:
In case anyone else makes the mistake I did: I also had this problem, but it was because I has used `apt` to install pygame instead of `pip`. Once I installed with `pip`, everything was fine.

OlivierDelbeke, 2020-04-21:
Indeed, installing pygame with pip instead of apt also solved my issue !