carla-simulator/carla #993: how to get offroad info of vehicles in new release from Python API

zj-gm, 2018-11-26:
hey, Carla team, I am trying to detect the offroad event of vehicles in simulator. is there a way to get from current Python APIs ? Thxs

nsubiron, 2018-12-04:
Hi @zj-gm,

At this point the event only tells you if you crossed a solid lane marking, that currently happens only if you either go off-road (to the right of your lane) or to the opposite lane (to the left of your lane).

closed 2019-02-09 by stale[bot]

noahcao, 2019-08-13:
Hi @nsubiron ,
I can get the lane type of a waypoint from carla 0.9.4. I tried to detect offroad by checking whether current vehicle's waypoint is of "driving" lane type. But there is some issue that the lane type is always "driving" even if the vehicle is on the garden grass. Can you provide a solution to it or any other solution to detect offroad? Thank you.

Rustypp, 2020-09-01:
Vibration increases as you leave the road, constantly monitor vibrations then look for changes?