carla-simulator/carla #995: Configur lidar camera in 0.9.1

bigsheep2018, 2018-11-27:
Hello, 

I build the source of 0.9.1 and run the manual_control.py.
I tried to save the point clouds to my disk and found the point cloud is not 360-degree covered.

Like:
[image]

May I know how/where to configure the lidar camera such that it could cover the entire 360-degree environment?

And please tell me how to obtain the 3d bounding box or 2d bev bounding box of different objects(e.g., car)

nsubiron, 2018-11-27:
Hi @bigsheep2018,

Please check out #835.

The 3D bounding box can be retrieved from the vehicle actor in Python. For instance,
```py
for vehicle in world.get_actors().filter('vehicle.*'):
    print(vehicle.bounding_box)
```

bigsheep2018, 2018-11-27:
Hello @nsubiron ,

I have checked the #835 and figure out how to change the configuration in Deprecated/PythonClient/client_example.py and manual_control.py

however, the server crashed down when I tried to run these examples under Deprecated/PythonClient/.
I am wandering if there is any other examples that shows how to config the lidar camera. Or may I know the how the manual_control.py under PythonAPI/ config the lidar camera?

Btw, I can run the examples under PythonAPI/ without any errors.

Thanks in advance,
Lin

bigsheep2018, 2018-11-27:
I see. Thanks. This helps a lot.
One more question, is it possible to design my own lidar camera somewhere and maybe add it to this project?
For example, I want a lidar that I could set how the casting rays distribute or maybe have a lidar intensity computation formula.

nsubiron, 2018-11-27:
Yes, take a look at how the current lidar is implemented at [RayCastLidar.h](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/RayCastLidar.h). For a bit more details on how the sensor system works, there is a mini tutorial in #830.

bigsheep2018, 2018-11-29:
Update:
It is able to find the ray_castv2 now, I actually do not change anything. Maybe it is because my desktop is too slow and the compiling somehow delayed. 

But there is another error coming out when I try to run manual_control.py:
terminate called after throwing an instance of 'clmdep_msgpack::v1::type_error'
what():  std::bad_cast

This happens right after I try to get the lidar image from the server, namely, this function:
self.sensor.listen(lambda image: CameraManager._parse_image(weak_self, image))

As I copy the code from RayCastLider.h/.cpp, I am assuming the data serialization and deserialization should be fine for this case, which means the client side should display exact the same lidar image as the original lidar camera displays (I have two lidar camera with same configuration)

It seems there is something wrong when I try to get the lidar image from the second lidar camera.
Hope you could give me some hints.

Thanks.

nsubiron, 2018-11-29:
The msgpack serialization is only used for the RPC channel, I would say is not an issue of the serialization of the point cloud but something else.

Could you get a stack trace for the exception? You can run the client with GDB
```
gdb --args python manual_control.py
```
then `r` for running and `bt` for getting the backtrace.

bigsheep2018, 2018-12-03:
hello @nsubiron ,
sorry for reply late. The followings are what I get following your instructions:

terminate called after throwing an instance of 'clmdep_msgpack::v1::type_error'
  what():  std::bad_cast

Thread 11 "python" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffceffc700 (LWP 19649)]
0x00007ffff7825428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  0x00007ffff7825428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff782702a in __GI_abort () at abort.c:89
#2  0x00007ffff5552da5 in __gnu_cxx::__verbose_terminate_handler() ()
   from /home/yuliu/.cache/Python-Eggs/carla-0.9.1-py3.5-linux-x86_64.egg-tmp/carla/libcarla.cpython-35m-x86_64-linux-gnu.so
#3  0x00007ffff5545e86 in __cxxabiv1::__terminate(void (*)()) ()
   from /home/yuliu/.cache/Python-Eggs/carla-0.9.1-py3.5-linux-x86_64.egg-tmp/carla/libcarla.cpython-35m-x86_64-linux-gnu.so
#4  0x00007ffff5545ed1 in std::terminate() ()
   from /home/yuliu/.cache/Python-Eggs/carla-0.9.1-py3.5-linux-x86_64.egg-tmp/carla/libcarla.cpython-35m-x86_64-linux-gnu.so
#5  0x00007ffff5572ebe in execute_native_thread_routine ()
   from /home/yuliu/.cache/Python-Eggs/carla-0.9.1-py3.5-linux-x86_64.egg-tmp/carla/libcarla.cpython-35m-x86_64-linux-gnu.so
#6  0x00007ffff7bc16ba in start_thread (arg=0x7fffceffc700) at pthread_create.c:333
#7  0x00007ffff78f741d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

nsubiron, 2018-12-03:
I see, that doesn't give much info. Can you try to catch the exception? with `catch throw`.

The set function, `void ARayCastLidar::Set(const FActorDescription &ActorDescription)`, is called when the lidar is created from the client-side. That function calls CreateLasers().

bigsheep2018, 2018-12-06:
I may possibly work on creating another camera later.
May I know how to transfer world coordinates into lidar camera's local space?
or How to get the bounding box with center in lidar's local space?

Thanks

closed 2019-02-15 by stale[bot]

kartik144, 2019-04-02:
> The files under "Deprecated" folder are not compatible with the last version, we're just keeping them until we port everything to the new API. You should see the examples in "PythonAPI".
> 
> In the new API you can adjust the frequency in the lidar blueprint
> 
> ```python
> bp = blueprint_library.find('sensor.lidar.ray_cast')
> bp.set_attribute('rotation_frequency', '10')
> ```

Hi, can you tell what all attributes are present in the LIDAR sensor blueprint? I want to capture LIDAR in a frame similar to other sensors, i.e. through a dashboard camera for point cloud registration between the LIDAR points and semantic segmentation images. Can you tell me how do I have to configure my LIDAR sensor for that?

visakhchikku, 2023-10-05:
Is it possible to save lidar data as .pcap file using carla?