carla-simulator/carla #996: [0.9.1] any suggestion about generating map with opendrive file?

hdy117, 2018-11-27:
I want to generate map in carla according to a opendrive file.
I search the plugin and find an "OpenDriveActor". This actor will load opendrive file.
I want to know when this actor will work?
Also, any suggestion about generating map automatically?
thank you very much.

marcgpuig, 2018-11-27:
Hi @hdy117 

The `OpenDriveActor` does not generate the map itself, the map and the OpenDrive, both, are generated using [RoadRunner](https://www.vectorzero.io/).

matthew-rococo, 2018-11-29:
i have contact with roadrunner, and use the trial version made a simple road map, so how can import this map to carla? 
i export the road map as fbx format, import it to carla, but the spawned vehicle just go directly , not follow the road.

marcgpuig, 2018-11-29:
Hi @matthew-rococo.
The @carla-simulator/art team is working on the documentation about this topic.

matthew-rococo, 2018-11-30:
Good news, very appreciating of your work !

hdy117, 2018-11-30:
> Hi @matthew-rococo.
> The @carla-simulator/art team is working on the documentation about this topic.

Now, if I place a opendrive file under /game/carla/maps/opendrive. Carla will build road network and route planner according to this file if I also create a level which share the same name. Am I right?
If yes, then I think I can just place a big flat floor in UE4, I don't have to build road network using roadrunner. Am I right?
If yes, Then I can use carla to test trajectory algorithm on a custom map with minimum build effort.

XGodina, 2018-11-30:
Hi Guys !!! The main steps RoadRunner - Unreal with Opendrive

When you created the map in RoadRunner. Export the map FBX. and OpenDrive.
FBX import into Unreal and Opendrive in file under /game/carla/maps/opendrive. "IMPORTANT" same name Map and OpenDrive. 

Once this is done, go to Unreal and drop on the map the actor "OpenDriveActor". This Actor generate automatically the opendrive.
After that, place an actor "Vehicle Spawn Point" over any OpenDrive generated trigger.
And Play !!!

Soon we will publish the documentation with the most detailed steps.

hdy117, 2018-12-04:
@XGodina I have done as your instruction.
I have my own xodr file, and I create a empyt level  with the same name.
[image]

When I run python client I got some error:

Hello from the pygame community. https://www.pygame.org/contribute.html
INFO: listening to server 127.0.0.1:2000

Welcome to CARLA manual control.

Use ARROWS or WASD keys for control.

    W            : throttle
    S            : brake
    AD           : steer
    Q            : toggle reverse
    Space        : hand-brake
    P            : toggle autopilot

    TAB          : change sensor position
    `            : next sensor
    [1-9]        : change to sensor [1-9]
    C            : change weather (Shift+C reverse)
    Backspace    : change vehicle

    R            : toggle recording images to disk

    F1           : toggle HUD
    H/?          : toggle help
    ESC          : quit

ERROR: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
Traceback (most recent call last):
  File "manual_control.py", line 695, in main
    game_loop(args)
  File "manual_control.py", line 629, in game_loop
    world = World(client.get_world(), hud)
  File "manual_control.py", line 134, in __init__
    spawn_points = self.world.get_map().get_spawn_points()
IndexError: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)

XGodina, 2018-12-05:
Hi David,
Dou you put the actor "Vehicle Spawn Point" ? .This actor's function is to spawn vehicles. The OpenDrive generates triggers. You only drag the "Vehicle Spawn Point" above these. I hope this solve your issue.

hdy117, 2018-12-06:
> Hi David,
> Dou you put the actor "Vehicle Spawn Point" ? .This actor's function is to spawn vehicles. The OpenDrive generates triggers. You only drag the "Vehicle Spawn Point" above these. I hope this solve your issue.

Yes, I did, as you can see in the picture. I have dragged 1 "Vehicle Spawn Point. I als0 test 2 or 3 vehicle spawn point, but it makes no difference

closed 2019-02-11 by stale[bot]

ShanshanYOU, 2019-05-02:
Is there any possibility to import directly maps created in RoadRunner into Carla, without passing by Unreal Engine ?

XGodina, 2019-05-03:
Currently no, we have many things that depend on unreal.

ShanshanYOU, 2019-05-03:
> Currently no, we have many things that depend on unreal.

I read some of your responses for the steps to import map created in RoadRunner into Carla Simulator, but I'm wondering which version of Carla is needed ? I cloned the "stable" and "master" branches, but didn't find "***/Carla/Map/Game...".  Could you plz describe a little bit in detail, how to export a map from  RoadRunner, than pass by UE4, and import into Carla ? 

It has been severals months when this issue is up, I'm wondering as well do you or your team have already a documentation about it ? Cuz I didn't find yet anyone has reached successfully to this issue with a documentation prepared.

Thanks a lot !

XGodina, 2019-05-03:
Hi, @ShanshanYOU 

If it is true that the documentation is needed an update and there are many things to consider to export the map. 

Currently, it has three-way to export maps

1- A script to automatically generate a map from RoadRunner (work in Linux version 0.9.1). https://github.com/carla-simulator/carla/blob/master/Docs/generate_map_from_fbx.md
2- Unreal Plugin from RoadRunner - https://www.youtube.com/watch?v=OcbYuTFAo8o
3- This is the manual method https://carla.readthedocs.io/en/latest/how_to_make_a_new_map/

Sorry, the way 1 it's not necessary to use Unreal. Export directly.

Atikul789, 2019-05-04:
How can I install opendrive on windows 10 any idea

Atikul789, 2019-05-04:
please

ShanshanYOU, 2019-05-15:
> Hi, @ShanshanYOU
>
> If it is true that the documentation is needed an update and there are many things to consider to export the map.
>
> Currently, it has three-way to export maps
>
> 1- A script to [...]

Hi, @XGodina Thank you for your reply. Actually I tried all these three ways, none of them worked well for me. The first one shows an error : "No smoothing group information". Equally for the second one, UE Editor couldn't find RoadRunnerImporter Plugin (which is contained in Plugins that I have copied).

Meantime, the third one doesn't provide a clear documentation. I'm in the mid way of it, could you clarify which meshs/elements of map should be put in each folder : Road/Roadlines/Terrain. I just have some meterials for road markings, roads/grass and trees. 

Thanks again ! 
Best regards,
Shanshan 

> [ ](https://guides.github.com/features/mastering-markdown/)

MatthiasThDs, 2019-05-22:
Hello,

does the "OpenDriveActor" generate the road geometry (mesh) from OpenDrive format? Or do i also need the fbx* file to see a geometry?

I was wondering, because in the "How to make a new map in RoadRunner" tutorial it sais i should export both filetypes.

Best regards

arvindpalkyutech, 2019-07-04:
Can I make a Map in CARLA without RoadRunner just using opendrive file. Actually I want to make same Map in CARLA as in OpenStreetMap. I can export OpenStreetMap to Opendrive format using SUMO simulator. Can Anyone guide me how to make fbx file and use it to create MAP in CARLA.

XGodina, 2019-07-04:
Hi, @arvindpalkyutech. I haven't used SUMO. If you can export .fbx file from SUMO, you can load in UE4. 
Another way is to load your OpenDrive in RoadRunner. Generate the map and after export in UE4.

jeffchen006, 2020-04-12:
Hi, I just meet a problem when using the Open Drive Actor. I have a umap file in Content Browser, but the box named "Generate Route" cannot be clicked. I'm so confused so any advise is welcomed.

marcgpuig, 2020-04-28:
@jeffchen006 Open Drive Actor is obsolete and will be removed soon, there is no need to use it since the new Trafic Manager works using the OpenDRIVE directly :)

ZJH166018, 2021-06-24:
@marcgpuig  Hi, I am doing co-simulation with Carla and sumo, when I process the map in Carla, I cannot find the "Open Drive Actor" or "OpenDRIVE".

XGodina, 2021-06-30:
Hi @ZJH166018,

If I understood good you are working in UE4. In that case, here you can find the Blueprint of OpenDriveActor. 

[image]

Remember that the Map file and the OpenDrive file must have the same name.

ZJH166018, 2021-06-30:
OK，I will check it, thank you very much!

发自我的iPhone

koach87, 2022-02-14:
Is there any way to parse OpenDrive file in Carla? And I don't have fbx data. 
I'd like to generate map in carla without RoadRunner because I haven't  receive any reply from Mathworks.

Thank you.