carla-simulator/carla #998: Prepare a dataset for RL

AliBaheri, 2018-11-27:
Hi CARLA team!

My goal is to prepare a dataset to test a reinforcement algorithm. To achieve that goal it seems I can use two script `manual.py` or `client_example.py` in autopilot mode. I read in one issue that the autopilot mode is not a good option for data collection? Can someone please clarify why? 
If one need to use `manual.py` is it necessary to use joystick or logitech?

Thanks in advance.

felipecode, 2018-12-05:
Hey, what do you mean by dataset for RL ? Shouldnt you try to learn directly by interaction ?
I think the best way to perform data collection is to have an "oracle" a hardcoded programmed agent that has priveledge information and drive perfectly on the CARLA scenarios.

felipecode, 2018-12-07:
check this repository 
https://github.com/carla-simulator/data-collector

felipecode, 2018-12-17:
It does not have bounding boxes on the image data directly. However it comes with semantic segmentation images and the world position for all the objects.

closed 2019-02-22 by stale[bot]