carla-simulator/carla #1000: [0.9.1 & 0.9.0] still build error on win10

Atlunch, 2018-11-28:
"C:\carla91\carla\Unreal\CarlaUE4\CarlaUE4.sln" (Standardziel) (1) ->
       "C:\carla91\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj" (Standardziel) (2) ->
       (Build Ziel) ->
         c:\program files\epic games\ue_4.19\engine\source\runtime\rhi\public\RHIResources.h(59): error C2039: "_Interl
       ockedCompareExchange": Ist kein Element von "FWindowsPlatformAtomics" [C:\carla91\carla\Unreal\CarlaUE4\Intermed
       iate\ProjectFiles\CarlaUE4.vcxproj]
         c:\program files\epic games\ue_4.19\engine\source\runtime\rhi\public\RHIResources.h(59): error C2665: "_Interl
       ockedCompareExchange": Durch keine der 4 Überladungen konnten alle Argumenttypen konvertiert werden. [C:\carla91
       \carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
         c:\program files\epic games\ue_4.19\engine\source\runtime\rhi\public\RHIResources.h(78): error C2039: "__fasts
       torefence": Ist kein Element von "FWindowsPlatformMisc" [C:\carla91\carla\Unreal\CarlaUE4\Intermediate\ProjectFi
       les\CarlaUE4.vcxproj]
         c:\program files\epic games\ue_4.19\engine\source\runtime\rhi\public\RHICommandList.h(282): error C2039: "_Int
       erlockedIncrement": Ist kein Element von "FWindowsPlatformAtomics" [C:\carla91\carla\Unreal\CarlaUE4\Intermediat
       e\ProjectFiles\CarlaUE4.vcxproj]
         c:\program files\epic games\ue_4.19\engine\source\runtime\rhi\public\RHICommandList.h(282): error C2665: "_Int
       erlockedIncrement": Durch keine der 4 Überladungen konnten alle Argumenttypen konvertiert werden. [C:\carla91\ca
       rla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(1689): error C2039: "G
       etObjectW": Ist kein Element von "TScriptInterface<IBlendableInterface>" [C:\carla91\carla\Unreal\CarlaUE4\Inter
       mediate\ProjectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(1709): error C2039: "G
       etObjectW": Ist kein Element von "TScriptInterface<IBlendableInterface>" [C:\carla91\carla\Unreal\CarlaUE4\Inter
       mediate\ProjectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(37): er
       ror C2660: "GetObjectW": Funktion akzeptiert keine 0 Argumente [C:\carla91\carla\Unreal\CarlaUE4\Intermediate\Pr
       ojectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(45): er
       ror C2660: "GetObjectW": Funktion akzeptiert keine 0 Argumente [C:\carla91\carla\Unreal\CarlaUE4\Intermediate\Pr
       ojectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(128): e
       rror C2660: "GetObjectW": Funktion akzeptiert keine 0 Argumente [C:\carla91\carla\Unreal\CarlaUE4\Intermediate\P
       rojectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(137): e
       rror C2660: "GetObjectW": Funktion akzeptiert keine 0 Argumente [C:\carla91\carla\Unreal\CarlaUE4\Intermediate\P
       rojectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(37): er
       ror C2660: "GetObjectW": Funktion akzeptiert keine 0 Argumente [C:\carla91\carla\Unreal\CarlaUE4\Intermediate\Pr
       ojectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(45): er
       ror C2660: "GetObjectW": Funktion akzeptiert keine 0 Argumente [C:\carla91\carla\Unreal\CarlaUE4\Intermediate\Pr
       ojectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(128): e
       rror C2660: "GetObjectW": Funktion akzeptiert keine 0 Argumente [C:\carla91\carla\Unreal\CarlaUE4\Intermediate\P
       rojectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(137): e
       rror C2660: "GetObjectW": Funktion akzeptiert keine 0 Argumente [C:\carla91\carla\Unreal\CarlaUE4\Intermediate\P
       rojectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\Core\Public\Async/AsyncFileHandle.h(103): error C203
       9: "__faststorefence": Ist kein Element von "FWindowsPlatformMisc" [C:\carla91\carla\Unreal\CarlaUE4\Intermediat
       e\ProjectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\Core\Public\Async/AsyncFileHandle.h(154): error C203
       9: "__faststorefence": Ist kein Element von "FWindowsPlatformMisc" [C:\carla91\carla\Unreal\CarlaUE4\Intermediat
       e\ProjectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\Core\Public\Async/AsyncFileHandle.h(159): error C203
       9: "__faststorefence": Ist kein Element von "FWindowsPlatformMisc" [C:\carla91\carla\Unreal\CarlaUE4\Intermediat
       e\ProjectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\Core\Public\Async/AsyncFileHandle.h(165): error C203
       9: "__faststorefence": Ist kein Element von "FWindowsPlatformMisc" [C:\carla91\carla\Unreal\CarlaUE4\Intermediat
       e\ProjectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(1689): error C2039: "G
       etObjectW": Ist kein Element von "TScriptInterface<IBlendableInterface>" [C:\carla91\carla\Unreal\CarlaUE4\Inter
       mediate\ProjectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\Engine\Classes\Engine/Scene.h(1709): error C2039: "G
       etObjectW": Ist kein Element von "TScriptInterface<IBlendableInterface>" [C:\carla91\carla\Unreal\CarlaUE4\Inter
       mediate\ProjectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(37): er
       ror C2660: "GetObjectW": Funktion akzeptiert keine 0 Argumente [C:\carla91\carla\Unreal\CarlaUE4\Intermediate\Pr
       ojectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(45): er
       ror C2660: "GetObjectW": Funktion akzeptiert keine 0 Argumente [C:\carla91\carla\Unreal\CarlaUE4\Intermediate\Pr
       ojectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(128): e
       rror C2660: "GetObjectW": Funktion akzeptiert keine 0 Argumente [C:\carla91\carla\Unreal\CarlaUE4\Intermediate\P
       rojectFiles\CarlaUE4.vcxproj]
         C:\Program Files\Epic Games\UE_4.19\Engine\Source\Runtime\CoreUObject\Public\UObject/UObjectIterator.h(137): e
       rror C2660: "GetObjectW": Funktion akzeptiert keine 0 Argumente [C:\carla91\carla\Unreal\CarlaUE4\Intermediate\P
       rojectFiles\CarlaUE4.vcxproj]
         UnrealBuildTool : error : UBT ERROR: Failed to produce item: C:\carla91\carla\Unreal\CarlaUE4\Plugins\Carla\Bi
       naries\Win64\UE4-Carla.lib [C:\carla91\carla\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]
         C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\Common7\IDE\VC\VCTargets\Microsoft.MakeFile.Targ
       ets(44,5): error MSB3075: Der Befehl ""C:\Program Files\Epic Games\UE_4.19\Engine\Build\BatchFiles\Build.bat" Ca
       rlaUE4 Win64 Development "C:\carla91\carla\Unreal\CarlaUE4\CarlaUE4.uproject" -WaitMutex -FromMsBuild" wurde mit
        dem Code 5 beendet. Stellen Sie sicher, dass Sie zur Ausführung dieses Befehls berechtigt sind. [C:\carla91\car
       la\Unreal\CarlaUE4\Intermediate\ProjectFiles\CarlaUE4.vcxproj]

Still getting these build errors.
I read in Fix win32 compilation #988 that you have changed something.
But still, I'm getting these build errors on win10.
What shall I do?

f5maki, 2018-11-29:
I have the same issue. 

The problem was fixed initially in the fix_win32_compilation branch from IFusionzz but carla.h was subsequently changed and the branch was deleted when merged to master.

If carla.h includes EngineMinimal.h then the windows build compiles. 
e.g. 
#include "Util/NonCopyable.h"
#include "EngineMinimal.h"
#include "Logging/LogMacros.h"
#include "Modules/ModuleInterface.h"

dieteekanne2, 2018-11-29:
> I have the same issue.
>
> The problem was fixed initially in the fix_win32_compilation branch from IFusionzz but carla.h was subsequently changed and the branch was deleted when merged to master.
>
> If [...]

Hey @f5maki could you solve this problem with including EngineMinimal.h?

marcgpuig, 2018-11-30:
@Atlunch @dieteekanne2 

Faced the same problem in my personal machine. But found a solution, update your Visual Studio :)
(Without the changes that @f5maki proposed works for me)

#### How to update Visual Studio Community 2017
1. Run the `Visual Studio Installer` (you can search it on the windows start menu)
1. Follow the instructions to update `Visual Studio Community 2017`

mvgraf, 2018-12-04:
> Hey @f5maki could you solve this problem with including EngineMinimal.h?

Worked perfectly. Thanks!

yyuuliang, 2018-12-06:
There is a branch "fix_vs2017_build_solution", try this one.
I am using win10 too, this branch works.

dieteekanne2, 2018-12-06:
> 
> 
> There is a branch "fix_vs2017_build_solution", try this one.
> I am using win10 too, this branch works. It is 0.9.0. You have to manually run the setup.py in PythonAPI to build the PythonAPI. There is a folder PythonClient, this one doesn't compatible with the server, so don't use this.( just delete it for convenience)

Hey @yyuuliang thanks for your answer. I tried to build this branch, however I'm getting an error with the "b2.exe". Couldnt find anyting on GitHub regarding this error.

<details>
  <summary> Error message</summary>
   -[install_boost]: Building...
C:/Carla/carla_0.9/Fix_VS_2017/carla-fix_vs2017_build_solution/carla-fix_vs2017_build_solution/Build/boost-src/libs/predef/check/../tools/check\predef.jam:46: Unescaped special character in argument $(language)::$(expression)
Performing configuration checks

    - default address-model    : 32-bit
    - default architecture     : x86
    - symlinks supported       : yes
    - C++11 mutex              : yes
    - Boost.Config Feature Check: cxx11_auto_declarations : yes
    - Boost.Config Feature Check: cxx11_constexpr : yes
    - Boost.Config Feature Check: cxx11_defaulted_functions : yes
    - Boost.Config Feature Check: cxx11_final : yes
    - Boost.Config Feature Check: cxx11_hdr_mutex : yes
    - Boost.Config Feature Check: cxx11_hdr_regex : yes
    - Boost.Config Feature Check: cxx11_hdr_tuple : yes
    - Boost.Config Feature Check: cxx11_lambdas : yes
    - Boost.Config Feature Check: cxx11_noexcept : yes
    - Boost.Config Feature Check: cxx11_nullptr : yes
    - Boost.Config Feature Check: cxx11_rvalue_references : yes
    - Boost.Config Feature Check: cxx11_template_aliases : yes
    - Boost.Config Feature Check: cxx11_thread_local : yes
    - Boost.Config Feature Check: cxx11_variadic_templates : yes
    - has_icu builds           : no
warning: Graph library does not contain MPI-based parallel components.
note: to enable them, add "using mpi ;" to your user-config.jam
    - zlib                     : no
    - bzip2                    : no
    - lzma                     : no
    - iconv (libc)             : no
    - iconv (separate)         : no
    - icu                      : no
    - icu (lib64)              : no
    - native-atomic-int32-supported : yes
    - message-compiler         : yes
    - native-syslog-supported  : no
    - pthread-supports-robust-mutexes : no
    - compiler-supports-visibility : no
    - compiler-supports-ssse3  : yes
    - compiler-supports-avx2   : yes
    - gcc visibility           : no
    - long double support      : yes
warning: skipping optional Message Passing Interface (MPI) library.
note: to enable MPI support, add "using mpi ;" to user-config.jam.
note: to suppress this message, pass "--without-mpi" to bjam.
note: otherwise, you can safely ignore this message.
    - libbacktrace builds      : no
    - addr2line builds         : no
    - WinDbg builds            : yes
    - WinDbgCached builds      : yes
failed to write output file 'build\boost\bin.v2\check\predef\msvc-14.1\debug\address-model-64\architecture-x86\threading-multi\predef_check_cc_09a06289f4560e2f553bb0cbbec71436.obj.rsp'!

    -[install_boost]: [B2 ERROR] An error ocurred while installing using "b2.exe".
    -[install_boost]: Exiting with error...

</details>

yyuuliang, 2018-12-06:
What command did you use?
It said "failed to write output file", so maybe you want to run it as administrator?

dieteekanne2, 2018-12-07:
> 
> 
> What command did you use?
> It said "failed to write output file", so maybe you want to run it as administrator?

Hey @yyuuliang 
thanks for your assistance,  it is highly appreciated. 
I did run the x64 command prompt as administrator. First I used the 'make launch' command and the error occured. But the same error occurs whatever command I use (e.g make launch-editor, make PythonAPI).  I'll a different computer and see how that goes. 
Have a geat  weekend!
I could not figure out what the b2.exe actually i

marcgpuig, 2018-12-10:
Hi @Atlunch @f5maki @mvgraf @dieteekanne2 @yyuuliang 

Due to the big amount of work and the difficulty of maintaining this project on both platforms, Carla on Windows is not **currently** working.

### Localized Issues
- `libpng.h` and other image libraries are used to store images, but only works on Linux.
  + **Workaround:** You can replace ImageIO.h with this:
    ```cpp
    #pragma once
    namespace carla {
    namespace image {
      class ImageIO {
      public:
        template <typename ImageT>
        static void ReadImage(const std::string &, ImageT &) {}
        template <typename ViewT>
        static std::string WriteView(std::string out_filename, const ViewT &) {
          return out_filename;
        }
      };
    } // namespace image
    } // namespace carla
    ```
    but you can't save images from python using our functions
  + **Possible solution:** use [stb_image.h](https://github.com/nothings/stb/blob/master/stb_image.h) and [stb_image_write.h](https://github.com/nothings/stb/blob/master/stb_image_write.h) instead.
    * **Possible drawbacks:** Impact on performance and compression quality.

- Boost's `Filesystem` is not compiling on windows.
  + Needs further investigation

- Since this PR #1008 Windows have other errors.
  + Needs further investigation

We will try to help you fixing this issues the most we can, but we can't focus on this right now.
Our most sincere apologies.

dieteekanne2, 2018-12-10:
> 
> 
> Hi @Atlunch @f5maki @mvgraf @dieteekanne2 @yyuuliang
> 
> Due to the big amount of work and the difficulty of maintaining this project on both platforms, Carla on Windows is not **currently** working.
> ### Localized Issues
> 
>     * `libpng.h` and other image libraries are used to store images, but only works on Linux.
>       
>       * **Workaround:** You can replace ImageIO.h with this:
>         ```c++
>         #pragma once
>         namespace carla {
>         namespace image {
>           class ImageIO {
>           public:
>             template <typename ImageT>
>             static void ReadImage(const std::string &, ImageT &) {}
>             template <typename ViewT>
>             static std::string WriteView(std::string out_filename, const ViewT &) {
>               return out_filename;
>             }
>           };
>         } // namespace image
>         } // namespace carla
>         ```
>         
>         
>         but you can't save images from python using our functions.
>       * **Possible solution:** use [stb_image.h](https://github.com/nothings/stb/blob/master/stb_image.h) and [stb_image_write.h](https://github.com/nothings/stb/blob/master/stb_image_write.h) instead.
>         
>         * **Possible drawbacks:** Impact on performance and compression quality.
> 
>     * Boost's `Filesystem` is not compiling on windows.
>       
>       * Needs further investigation
> 
>     * Since this PR #1008 Windows have other errors.
>       
>       * Needs further investigation
> 
> 
> We will try to help you fixing this issues the most we can, but we can't focus on this right now.
> Our most sincere apologies.

Hello @marcgpuig 
thank you for the info!

eds89, 2019-01-24:
In the past, I've managed to compile CARLA 0.8.2 successfully on Windows 7 64-bit. Has this changed with CARLA 0.9.X?

PR #1241 (2019-02-27): Basic build system for Windows
#### Description
A basic build system for Windows.

**This PR provides**
- Support for Python3 (`make PythonAPI`)
- Automated download and build of `zlib` and `libpng` needed in Win
- Now Boost is downloaded from `https` and not from git

**This PR does NOT provides**
- Support for Python2 (not tested really)
- Support for Google Test

Fixes #1000 
Fixes #1155
Fixes #976

#### Full test of functionalities
1. Open a `Visual Studio x64 Native Tools Command Prompt` and navigate to the project folder
1. Remove `carla/Build` folder (if exists)
1. Run `make clean launc [... cut]
closed 2019-02-27 by nsubiron