carla-simulator/carla #1001: SetSimulatePhysics(true) results in some strange behavior for vehicle.carlamotors.carlacola

tanglrHello, 2018-11-29:
The first time I found this problem was when I spawned a vehicle.carlamotors.carlacola and then called ths SetSimulatePhysics(True), two of its four wheels dropped off immediately after the vehicle appeared in the scene. After I deleted the call to SetSimulatePhysics, everything goes well.

Then I tested it in manual_control.py.
1. Add a line in World.__init__(), after the self.world.spawn_actor() like the following:
`self.vehicle = self.world.spawn_actor(blueprint, spawn_point)
self.vehicle.set_simulate_physics(True)`
2. Choose "vehicle.carlamotors.carlacola" blueprint for vehicle in _get_random_blueprint:
`    def _get_random_blueprint(self):
        #bp = random.choice(self.world.get_blueprint_library().filter('vehicle'))
        bp = self.world.get_blueprint_library().find("vehicle.carlamotors.carlacola")
        if bp.has_attribute('color'):
            color = random.choice(bp.get_attribute('color').recommended_values)
            bp.set_attribute('color', color)
        return bp
`
3. Run manual_control.py. Then control the vehicle to move, and strange thing happens: all wheels won't move together with the vehicle body, and only the body are under control.

The carla version I'm using is release 0.9.1，and the unreal version is 4.19.

nsubiron, 2018-11-30:
Hi @tanglrHello,

Thanks for the report! it is indeed a bug that happens only with some of the vehicles.

nsubiron, 2018-12-21:
Fixed on latest update of the assets.

closed 2018-12-21 by nsubiron