carla-simulator/carla #1003: where is CarlaMapGenerator?

yyuuliang, 2018-11-29:
I am working on creating a new map. I tried to follow the instruction on the website, it says there is a "CarlaMapGenerator". But I don't know where to find it. I also googled this and it seems like everyone know this... But I can't find it :( I checked everywhere in my UE editor.I can see the Carla plugin installed. But where is the CarlaMapGenerator?

TheNihilisticRobot, 2018-11-30:
Hi @yyuuliang 

Our bad, we don't really have documented where CarlaMapGenerator is stored.

Unreal By default hides the C++ classes in the content browser so, to find it there you'll have first to open the "ViewOption" submenu and select "Show C++ Classes" Then you'll have access to new folders with our C++ classes.

Besides that, In the "Modes" window you can write the name of any asset you need and if it is in the project it will find it for you.

For the second question, the CarlaUE4.uproject file is created While building inside the project folder (Carla/Unreal/CarlaUE4). You could also write `make launch-only`  in the make console this will launch the editor with the previously built version of Carla.

yyuuliang, 2018-12-06:
@TheNihilisticRobot I checked the MapEditor, but now I don't know how to use it or if I need this. I am directly modifying the map in the Editor. Maybe later I will start to use this.
Thanks for your reply!

For the UE editor, I found that I can start the editor by double-clicking the project file in Carla/Unreal/CarlaUE4.

closed 2019-02-11 by stale[bot]

ShanshanYOU, 2019-05-15:
> Hi @yyuuliang
>
> Our bad, we don't really have documented where CarlaMapGenerator is stored.
>
> Unreal By default hides the C++ classes in the content browser so, to find it there you'll have first to [...]

Hello ! Still not found ! "CarlaMapGenerator" & "OpenDriveActor" are not in my list of modes. Where I could find "ViewOptions" ?

Cheers !