carla-simulator/carla #1004: How to use the PythonAPI in Carla 0.8.2, the stable version.

Johnnyswong, 2018-11-30:
HI, @sunpochin @harlowja @juaxix @iFuSiiOnzZ @laurent-george 
How to use the PythonAPI in Carla 0.8.2, Carla 0.9.1 has lots of awesome new features!

nsubiron, 2018-11-30:
The APIs are not compatible, precisely for all these new features :wink:

Please, do not @mention random people on your questions, we'll assign the right person to each issue.

Johnnyswong, 2018-12-03:
OK, I'm sorry for the random @mention. 
The scripts of manual_control.py, tutorial.py, and spawn_npc.py in the directory run smoothly, but when I run the code in imitation learning repo, the run_CIL.py script, it always report connection lost error.
BTW, the imitation learning model is training with carla version 0.8.2, I think provide API is a good choice for the users.

Johnnyswong, 2018-12-03:
Do you change the previous PythonClient library written in Python into PythonAPI written in Cpp? @nsubiron

Johnnyswong, 2018-12-03:
Thx @nsubiron 
Sorry for my random @mention. 
It's a pity that the imitation learning repo is not compatible with the new 0.9.1 version but with 0.8.2. Can you provide a compatible version?

closed 2019-02-09 by stale[bot]