carla-simulator/carla #1005: How can I know whether a transform is inside or outside the world bound?

tanglrHello, 2018-11-30:
If I spawn a vehicle at somewhere outside the world, the actor returned seem to be valid. But after about 1 minute later, the carla server crashes. Before crashing, a warning info is given: "[2018.11.30-02.37.31:353][221]LogActor: Warning: CarlaCola_C_2 is outside the world bounds!".

I searched the code within LibCarla, but nothing was found related to "outside the world bounds". So I guess this is a warn info given by unreal instead of carla. Maybe it's not dealed properly in carla for spawning outside the world, and carla just construct an Actor object and return, but something goes wrong somewhere.

So what should I do to avoid such corruption? Besides getting spawn points from carla and spawn at these points directly, how can I tell whether a transform is inside or outside the world bound?

marcgpuig, 2018-11-30:
Hi @tanglrHello
Seems like you are spawning (or moving) the vehicle far far away from the (0,0,0) of the world and this causes this kind of issue in Unreal Engine.
Please, read carefully your code and search for such behaviour.

nsubiron, 2018-11-30:
At the moment we don't have any mechanism for telling whether a point is inside the world bounds, but you can project any arbitrary location to a road by using waypoints.
```py
map = world.get_map()
wp = map.get_waypoint(location)
spawn_point = wp.transform
```

closed 2019-02-05 by stale[bot]