carla-simulator/carla #1006: Carla  0.91 how to move my viewpoint in the map along with my player vehicle??

anshulpaigwar, 2018-11-30:
Carla 0.91 has loads of useful features that can be used for research, thanks to the carla team. 
But unlike in Carla 0.8x versions, my viewpoint in the map does not move with the player vehicle. We have to manually move it with arrow keys and mouse.

[image]

I want to be able to see the environment around my player vehicle and what it is doing. I understand that this is not the case in Carla 0.9x due to multi-client support and we would have to do this using something like pygame. But it would be nice if we could directly shift viewpoints over different player vehicles in Carla window as in 0.8x version.
[image]

It would  be helpful if I could get the sample pygame code for viewing the camera outputs as in your video of Carla 9.0

M-Hammod, 2018-11-30:
you easily can do that  by piloting the camera attached to the vehicle, after spawning the vehicle from python, in the unreal editor-->world outliner-->search for 'cam'-->you will see the a camera actor attached to the vehicle that you spawned-->rightclick on it and choose pilot.

I had a proplem that I had to eject the player for the pilot option to get activated

hope this helps

anshulpaigwar, 2018-11-30:
Thanks for the reply it helped to solve my problem.

anshulpaigwar, 2018-12-04:
How to attach spectator to the vehicle like a camera ?? I tried this simply

```
        while True:
            spectator_transform =  vehicle.get_transform()
            spectator_transform.location += carla.Location(x = -2, y=0, z = 2.0)
            spectator.set_transform(spectator_transform)
            time.sleep(0.1)
```
This kind of works but there are two problems:
1)How to define transform with respect to vehicle?  According to my code the location of the spectator is sometimes ahead of the Vehicle sometimes behind. Also the x and y axis being getting swapped, I think it is because I have not considered orientation of the vehicle.

2) As I am using while loop with sleep, the update of spectator transform is not synced properly with the vehicle movement and it differs according to the processing power of different computers. If I reduce the sleep time the updates are fast with jitter and code freezes my computer.

nsubiron, 2018-12-04:
Unfortunately, the transformation matrices are not yet implemented in the new API, but you can get them in Python from the old API at [transform.py](https://github.com/carla-simulator/carla/blob/master/Deprecated/PythonClient/carla/transform.py).

The best way for synchronizing the client with simulator updates is using the wait for tick function
```py
timestamp = world.wait_for_tick()
delta_seconds = timestamp.delta_seconds
```

Sakrust, 2018-12-05:
Just use a little trick by placing a dummy actor in the scene: 

camera_bp = world.get_blueprint_library().find('sensor.camera.rgb')
    camera_transform = carla.Transform(carla.Location(x=10, z=10))
    camera = world.spawn_actor(camera_bp, camera_transform, attach_to=vehicle)
spectator.set_transform(camera.get_transform())

nsubiron, 2018-12-05:
That's a nice workaround :D but I would use instead a cheaper sensor, collision sensor for instance, cameras make your FPS drop significantly.

closed 2019-02-11 by stale[bot]

ChrisHRZ, 2019-04-17:
> Just use a little trick by placing a dummy actor in the scene:
>
> camera_bp = world.get_blueprint_library().find('sensor.camera.rgb')
> camera_transform = carla.Transform(carla.Location(x=10, z=10)) [...]

> Just use a little trick by placing a dummy actor in the scene:
>
> camera_bp = world.get_blueprint_library().find('sensor.camera.rgb')
> camera_transform = carla.Transform(carla.Location(x=10, z=10)) [...]

How can i chose to view this camera?

Sakrust, 2019-04-17:
You don't. Your simulator window view will be updated.

gunki-geek, 2022-01-25:
u can write two scripts master.py and slave.py and synchronize between the two like in the attached scripts
[attach_spectator_to_vehicle.zip](https://github.com/carla-simulator/carla/files/7932460/attach_spectator_to_vehicle.zip)
s

kuangzh123, 2024-03-02:
> Just use a little trick by placing a dummy actor in the scene:
>
> camera_bp = world.get_blueprint_library().find('sensor.camera.rgb') camera_transform = carla.Transform(carla.Location(x=10, z=10)) [...]

Does this work?When I use this method,the spectator still moves forward and backward as the car speeds up and down