carla-simulator/carla #1011: Release 0.9.1 manual_control.py gives error when run using python3.6 interpreter

YashBansod, 2018-12-04:
If using a python3.6 interpreter while running the example codes provided with the pre-compiled binaries, the appropriate path is not appended. This happens due to this code:
https://github.com/carla-simulator/carla/blob/bc5a3edd3d3a7a00a157cbe67d945da5a6d8fa53/PythonAPI/manual_control.py#L49-L55
The problem is caused by `sys.version_info.minor` which returns 6 as output but, only **carla-0.9.1-py2.7-linux-x86_64.egg** and **carla-0.9.1-py3.5-linux-x86_64.egg** files are present.
The problem resolves if you make a copy of the **carla-0.9.1-py3.5-linux-x86_64.egg** file and rename it to **carla-0.9.1-py3.6-linux-x86_64.egg**. 

However, I think that the code should be changed to something like: 
```python
sys.path.append(glob.glob('**/carla-*%d.*-%s.egg' % 
(sys.version_info.major, 'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
```
which would make running the code in Python3.x possible.

nsubiron, 2018-12-04:
Hi @YashBansod,

The way to solve this is actually to recompile the carla module for 3.6. Although it may work, an egg file compiled for Python 3.5 is not necessarily compatible with Python 3.6.

Atlunch, 2018-12-04:
Hello,
I just downloaded and used the pre-compiled version on linux for the first time.
I can start the server and let spawn a few npcs.
While I'm using the manual.control.py I'm getting a blackscreen, but i can control the car.
Is this the same issue like @YashBansod mentioned?

@nsubiron what do you mean with "recompile the carla module"? Which module and how can i recompile it?

[image]

nsubiron, 2018-12-05:
The Python module (PythonAPI) is compiled for a specific version of Python. This is done with `make PythonAPI`. Currently we compile for `/usr/bin/python2` and `/usr/bin/python3`, whatever those versions are in your machine (in Ubuntu usually 2.7 and 3.5). There is a PR (#949) to allow using the Python in your environment, but we're having some difficulties with it.

@Atlunch That black screen is probably related to the errors in pygame, I haven't seen that before. Which Python version are you using?

Atlunch, 2018-12-05:
@nsubiron 
I just did a full fresh reinstallation of my ubuntu 18.04.
I found out that i have the them issue like in #991 mentioned.
After the new build i have only python3 installed.
And I'm using the pre-compiled version. 
Starting the server works fine.
When I'm trying now starting "python3 manual_control.py", I'm getting this Error:

Traceback (most recent call last):
  File "manual_control.py", line 63, in <module>
    import carla
ModuleNotFoundError: No module named 'carla'

My current python3 version is Python 3.6.7.

Atlunch, 2018-12-05:
JUUUUHHUUUUUUUU.......I could solve my problem :dancer: 

I'm trying now to describe the whole process.

As I already mentioned, I did a fresh install of ubuntu 18.04.
I download the compiled version of 0.9.1.
I extracted everything into a new folder.
After this I started the server by typing in ./CarlaUE4.sh
The server worked fine, but i ran into the problem that trying to start "manual_control.py" by typing in "python3 manual_control.py" occured the error: No module named carla.

To solve this I stepped into the folder "PythonAPI" an extrated the package "carla-0.9.1-py3.5-linux-x86_64.egg". That the PythonAPI folder structure looked like this:
PythonAPI
- carla
- EGG-INFO

After that I had to copy the "manual_control.py" module into the "PythonAPI" folder.
After that I had to install pip,pygame and numpy by using this commands.
pip: sudo apt-get install python3-pip 
pygame: python3 -m pip install -U pygame --user
numpy: sudo apt-get install python3-numpy 

After all these steps i was able to run the "manual_control.py" out of the PythonAPI folder.

Thank you very much for your great work. #CarlaTeam

YashBansod, 2018-12-05:
@Atlunch thats another way to go about it. You could have referred the #985 to save some time.
https://github.com/carla-simulator/carla/issues/985#issuecomment-440917810

Also I wanted to know what problem you were facing with this solution: 
https://github.com/carla-simulator/carla/issues/1011#issue-387178266

Atlunch, 2018-12-05:
@YashBansod ah i see you already found a way. I overlooked it.

First I had the problem getting a black screen while using manual_control.py.
But I think that this problem occured because I used a wrong python and pygame version.
After a fresh install I had the same issue like you...no carla module

HubFire, 2018-12-11:
Carla 0.9.1 supports both Python2 and Python3

1. install Anaconda and create python2.7 and python3.7：
`conda create -n py37 python=3.7 `
add another python:
`conda create -n py27 python=2.7 `

2. modefy  setup.sh  in  Util/BuildTools

  py2="~/anaconda3/envs/py27/python"
  py2_root=`${py2} -c "import sys; print(sys.prefix)"`
  pyv=`$py2 -c "import sys;x='{v[0]}.{v[1]}'.format(v=list(sys.version_info[:2]));sys.stdout.write(x)";`
  ./bootstrap.sh \
      --with-toolset=clang \
      --prefix=../boost-install \
      --with-libraries=python,filesystem \
      --with-python=${py2} --with-python-root=${py2_root}

  if ${TRAVIS}
  then
    echo "using python : ${pyv} : ${py2_root}/bin/python2 ;" > ${HOME}/user-config.jam
  else
    echo "using python : ${pyv} : ${py2_root}/bin/python2 ;" > project-config.jam
  fi

  ./b2 toolset="${BOOST_TOOLSET}" cxxflags="${BOOST_CFLAGS}" --prefix="../${BOOST_BASENAME}-install" -j ${CARLA_BUILD_CONCURRENCY} stage release
  ./b2 toolset="${BOOST_TOOLSET}" cxxflags="${BOOST_CFLAGS}" --prefix="../${BOOST_BASENAME}-install" -j ${CARLA_BUILD_CONCURRENCY} install
  ./b2 toolset="${BOOST_TOOLSET}" cxxflags="${BOOST_CFLAGS}" --prefix="../${BOOST_BASENAME}-install" -j ${CARLA_BUILD_CONCURRENCY} --clean-all
  
  
  popd >/dev/null
  rm -Rf ${BOOST_BASENAME}-source
  tar -xzf boost_1_67_0.tar.gz
  mkdir -p ${BOOST_BASENAME}-install/include
  mv boost_1_67_0 ${BOOST_BASENAME}-source
  pushd ${BOOST_BASENAME}-source >/dev/null

  py3="~/anaconda3/envs/py37/python"
  py3_root=`${py3} -c "import sys; print(sys.prefix)"`
  pyv=`$py3 -c "import sys;x='{v[0]}.{v[1]}'.format(v=list(sys.version_info[:2]));sys.stdout.write(x)";`
  ./bootstrap.sh \
      --with-toolset=clang \
      --prefix=../boost-install \
      --with-libraries=python \
      --with-python=${py3} --with-python-root=${py3_root}

  if ${TRAVIS}
  then
    echo "using python : ${pyv} : ${py3_root}/bin/python3 ;" > ${HOME}/user-config.jam
  else
    echo "using python : ${pyv} : ${py3_root}/bin/python3 ;" > project-config.jam
  fi

  ./b2 toolset="${BOOST_TOOLSET}" cxxflags="${BOOST_CFLAGS}" --prefix="../${BOOST_BASENAME}-install" -j ${CARLA_BUILD_CONCURRENCY} stage release
  ./b2 toolset="${BOOST_TOOLSET}" cxxflags="${BOOST_CFLAGS}" --prefix="../${BOOST_BASENAME}-install" -j ${CARLA_BUILD_CONCURRENCY} install

this will  create  libboost_python2.7 and libboost_python3.7

3.  run  setup ，do not forget run  update.sh  before that
`export CPLUS__INCLUDE_PATH=/home/huyi/anaconda3/include/python3.7m`
`make setup`
4.`make rebuild`
5.`make package`
   make sure  the step 3,4,5 all  success.
   this will  create both 2.7 and 3.7 egg file in Dist/0.9.1-dirty/LinuxNoEditor/PythonAPI
6.install  carla python API ：
  python2.7 ：
 `source activate py27`
` cd  Dist/0.9.1-dirty/LinuxNoEditor/PythonAPI`
` easy_install  carla-xx-py2.7-xx.egg`
 python3.7 ：
 `source activate py37`
` cd  Dist/0.9.1-dirty/LinuxNoEditor/PythonAPI`
` easy_install  carla-xx-py3.7-xx.egg`

Now you can use python2 and python3 with carla 0.9.1.