carla-simulator/carla #1013: PythonAPI and boost errors

mvgraf, 2018-12-05:
Hi,

when using `make PythonAPI ` within the x64 console with Python 3.5 installed (as I guess is correct due to your documentation) using the 0.9.1 carla version (master branch), I always get weird boost errors:

`dependencies/include\boost/python/signature.hpp(236): note: see declaration of 'boost::python::detail::get_signature'
dependencies/include\boost/python/class.hpp(497): error C2780: 'boost::mpl::vector11<RT,ClassT&,T0,T1,T2,T3,T4,T5,T6,T7,T8> boost::python::detail::get
_signature(RT (__cdecl ClassT::* )(T0,T1,T2,T3,T4,T5,T6,T7,T8))': expects 1 arguments - 2 provided
dependencies/include\boost/python/signature.hpp(218): note: see declaration of 'boost::python::detail::get_signature'
dependencies/include\boost/python/class.hpp(497): error C2784: 'boost::mpl::vector10<T0,T1,T2,T3,T4,T5,T6,T7,T8,T9> boost::python::detail::get_signatu
re(RT (__cdecl *)(T0,T1,T2,T3,T4,T5,T6,T7,T8),void *)': could not deduce template argument for 'RT (__cdecl *)(T0,T1,T2,T3,T4,T5,T6,T7,T8)' from 'Fn'
        with
        [
            Fn=bool (__cdecl carla::client::ActorAttribute::* )(const bool &) const
        ]
dependencies/include\boost/python/signature.hpp(194): note: see declaration of 'boost::python::detail::get_signature'
dependencies/include\boost/python/class.hpp(497): error C2782: 'boost::mpl::vector10<RT,most_derived<Target,ClassT>::type&,T0,T1,T2,T3,T4,T5,T6,T7> bo
ost::python::detail::get_signature(RT (__cdecl ClassT::* )(T0,T1,T2,T3,T4,T5,T6,T7) volatile const,Target *)': template parameter 'ClassT' is ambiguou
s
dependencies/include\boost/python/signature.hpp(236): note: see declaration of 'boost::python::detail::get_signature'
dependencies/include\boost/python/class.hpp(497): note: could be 'carla::client::ActorAttributeValueAccess'
dependencies/include\boost/python/class.hpp(497): note: or       'carla::client::ActorAttribute'
dependencies/include\boost/python/class.hpp(497): error C2784: 'boost::mpl::vector10<RT,most_derived<Target,ClassT>::type&,T0,T1,T2,T3,T4,T5,T6,T7> bo
ost::python::detail::get_signature(RT (__cdecl ClassT::* )(T0,T1,T2,T3,T4,T5,T6,T7) volatile const,Target *)': could not deduce template argument for
'RT (__cdecl ClassT::* )(T0,T1,T2,T3,T4,T5,T6,T7) volatile const' from 'Fn'
        with
        [
            Fn=bool (__cdecl carla::client::ActorAttribute::* )(const bool &) const
        ]
dependencies/include\boost/python/signature.hpp(236): note: see declaration of 'boost::python::detail::get_signature'
dependencies/include\boost/python/class.hpp(497): error C2780: 'boost::mpl::vector10<RT,ClassT&,T0,T1,T2,T3,T4,T5,T6,T7> boost::python::detail::get_si
gnature(RT (__cdecl ClassT::* )(T0,T1,T2,T3,T4,T5,T6,T7) volatile const)': expects 1 arguments - 2 provided
dependencies/include\boost/python/signature.hpp(218): note: see declaration of 'boost::python::detail::get_signature'
dependencies/include\boost/python/class.hpp(497): error C2782: 'boost::mpl::vector10<RT,most_derived<Target,ClassT>::type&,T0,T1,T2,T3,T4,T5,T6,T7> bo
ost::python::detail::get_signature(RT (__cdecl ClassT::* )(T0,T1,T2,T3,T4,T5,T6,T7) volatile,Target *)': template parameter 'ClassT' is ambiguous
dependencies/include\boost/python/signature.hpp(236): note: see declaration of 'boost::python::detail::get_signature'
dependencies/include\boost/python/class.hpp(497): note: could be 'const carla::client::ActorAttributeValueAccess'
dependencies/include\boost/python/class.hpp(497): note: or       'carla::client::ActorAttribute'
dependencies/include\boost/python/class.hpp(497): error C2784: 'boost::mpl::vector10<RT,most_derived<Target,ClassT>::type&,T0,T1,T2,T3,T4,T5,T6,T7> bo
ost::python::detail::get_signature(RT (__cdecl ClassT::* )(T0,T1,T2,T3,T4,T5,T6,T7) volatile,Target *)': could not deduce template argument for 'RT (_
_cdecl ClassT::* )(T0,T1,T2,T3,T4,T5,T6,T7) volatile' from 'Fn'
        with
        [
            Fn=bool (__cdecl carla::client::ActorAttribute::* )(const bool &) const
        ]
dependencies/include\boost/python/signature.hpp(236): note: see declaration of 'boost::python::detail::get_signature'
dependencies/include\boost/python/class.hpp(497): error C2780: 'boost::mpl::vector10<RT,ClassT&,T0,T1,T2,T3,T4,T5,T6,T7> boost::python::detail::get_si
gnature(RT (__cdecl ClassT::* )(T0,T1,T2,T3,T4,T5,T6,T7) volatile)': expects 1 arguments - 2 provided
dependencies/include\boost/python/signature.hpp(218): note: see declaration of 'boost::python::detail::get_signature'
dependencies/include\boost/python/class.hpp(497): error C2782: 'boost::mpl::vector10<RT,most_derived<Target,ClassT>::type&,T0,T1,T2,T3,T4,T5,T6,T7> bo
ost::python::detail::get_signature(RT (__cdecl ClassT::* )(T0,T1,T2,T3,T4,T5,T6,T7) const,Target *)': template parameter 'ClassT' is ambiguous
dependencies/include\boost/python/signature.hpp(236): note: see declaration of 'boost::python::detail::get_signature'
dependencies/include\boost/python/class.hpp(497): note: could be 'carla::client::ActorAttributeValueAccess'
dependencies/include\boost/python/class.hpp(497): note: or       'carla::client::ActorAttribute'
dependencies/include\boost/python/class.hpp(497): error C2784: 'boost::mpl::vector10<RT,most_derived<Target,ClassT>::type&,T0,T1,T2,T3,T4,T5,T6,T7> bo
ost::python::detail::get_signature(RT (__cdecl ClassT::* )(T0,T1,T2,T3,T4,T5,T6,T7) const,Target *)': could not deduce template argument for 'RT (__cd
ecl ClassT::* )(T0,T1,T2,T3,T4,T5,T6,T7) const' from 'Fn'
        with
        [
            Fn=bool (__cdecl carla::client::ActorAttribute::* )(const bool &) const
        ]
dependencies/include\boost/python/signature.hpp(236): note: see declaration of 'boost::python::detail::get_signature'
dependencies/include\boost/python/class.hpp(497): error C2780: 'boost::mpl::vector10<RT,ClassT&,T0,T1,T2,T3,T4,T5,T6,T7> boost::python::detail::get_si
gnature(RT (__cdecl ClassT::* )(T0,T1,T2,T3,T4,T5,T6,T7) const)': expects 1 arguments - 2 provided
dependencies/include\boost/python/signature.hpp(218): note: see declaration of 'boost::python::detail::get_signature'
dependencies/include\boost/python/class.hpp(497): fatal error C1003: error count exceeds 100; stopping compilation
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2017\\Professional\\VC\\Tools\\MSVC\\14.16.27023\\bin\\HostX86\\x64\\cl.exe' failed
with exit status 2

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "D:/Projects/carla/PythonAPI\dist"!`

How is this possible, as boost is shipped within the carla dependencies installation?

Cheers,
Georg.

Atlunch, 2018-12-05:
Hello,
I just build also the PythonAPI by the command "make PythonAPI".
I'm using Ubuntu 18.04 and Python 3.6.7, it works fine.

mvgraf, 2018-12-07:
Yeah I'm building on Windows and it seems more of a Boost error :(

marcgpuig, 2018-12-10:
Let's continue this discusion on #1000

closed 2018-12-10 by marcgpuig