carla-simulator/carla #1015: [0.9.1] [Linux] Map customization instructions

Atlunch, 2018-12-05:
Hello everyone,
I just build a new carla 0.9.1 build from source.
Now I would like to customize my own map. I got also a licence for Roadrunner.
What I have now to do after I lunched the CarleUE4 project?
In the docs and in github I read something about duplicating a map and the CarlaMapGenerator.
How can I duplicate the map inside the editor and what i have exactly to do with the CarlaMapGenerator. I found the C++ class but i also read something about clicking "Trigger Road Map Generation". The C++ class is just grey and i can only open it with visual code.
But how can I use the Generator like i read?

I hope someone could give a much more detailed instruction.
A short tutorial video would be awesome.

yyuuliang, 2018-12-06:
What I am doing: directly use the UE editor to edit the map. Following the instruction in the document. then "make clean && make launch" to rebuild the binaries. 
It works so far so good.

Atlunch, 2018-12-06:
And what are you exactly doing with the CarlaMapGenerator?
I found only these C++ class like in this image. I can open it with visual code and there isn't something like seed oder Trigger Road Map Generator. 
How can I interact with it like in the documentation mentioned?

[image]

TheNihilisticRobot, 2018-12-06:
Hi @yyuuliang and @Atlunch We're currently working on documenting the Map generation pipeline. Please check Issue #996 While we get the documentation ready. There is a simplified set of instructions to create a map in roadrunner by @XGodina in there.

yyuuliang, 2018-12-06:
@Atlunch I didn't know how to use this CarlaMapGenerator, either...

XGodina, 2018-12-07:
Hi Guys !!! The main steps RoadRunner - Unreal with Opendrive

When you created the map in RoadRunner. Export the map FBX. and OpenDrive.
FBX import into Unreal and Opendrive in file under /game/carla/maps/opendrive. "IMPORTANT" same name Map and OpenDrive.

Once this is done, go to Unreal and drop on the map the actor "OpenDriveActor". This Actor generate automatically the opendrive.

 the current version of carla needs a CarlaMapGenerator Spawned in order for our vehicles to drive.
For the moment Create a CarlaMapGenerator away from your city, never above or under it  and build its navigation.

After that, place an actor "Vehicle Spawn Point" over any OpenDrive generated trigger.
And Play !!!

Soon we will publish the documentation with the most detailed steps.

matthew-rococo, 2018-12-08:
@XGodina i do exactly as your instruction, and everything is OK except that there is no  auto generated route planner node on the map. then when i spawned one car on the road using python client, the car will drive directly to the end of the map(not follow the road )!
what important things i have missed?

Atlunch, 2018-12-08:
@XGodina @matthew-rococo 
How could you import the OpenDrive xodr file?
I am not able to select it by using the import option of Unreal.

Atlunch, 2018-12-08:
Finally I made it!!!!!!!!!!!!!!!!!!!!!!!!!

My System:
Processor: i7-4790k
RAM: 16 GB
GPU: Nvidia GTX 980 Ti (Driver 410)
OS: Ubuntu 16.04 (I had to switch from 18 to 16 because Roadrunner is actually only running on 16)

[image]
[image]
[image]

My steps creating my own map using Roadrunner and Carla 0.9.1.

1. I downloaded the carla 0.9.1 source and build it like in the docs its mentioned.
Furthermore I build also the PythonAPI with make PythonAPI to try if the server and the client connections works.
2. Like my example I created just a circle road in roadrunner.
3. Export the FBX and OpenDrive using File -> Export ... you have to export both separatly.
4. Import the Roadrunner Plugin into the CarlaUE4 Project File following this tutorial.
https://www.youtube.com/watch?v=OcbYuTFAo8o
You will find the plugin folder (on Ubuntu) in /usr/bin/VectorZero/Tools/Unreal/Plugins.
IMPORTANT: you cant run the CarlaUE4 with make launch or launch-only. It wont work. You have to start just the UE4 Project after you copied the RoadrunnerImporter into the Projekt Pluging folder. Doing this you can install the Plugin into your project.
5. Now copy the exported OpenDrive xodr file into your carla project folder. Not with the Unreal Editor. The Unreal Editor cant read OpenDrive files xodr.
/home/yourname/carla91s/Unreal/CarlaUE4/Content/Carla/Maps/OpenDrive.
Name the File like @XGodina mentioned like you will name your Level later in the Editor.
6. Open the CarlaUE4 Project with make launch-only.
7.Now generate a new Level with File -> New Leve inside the Maps and name it like the OpenDrive file.
8. Import the FBX file into a separatly folder in the content folder. Doing this it will open a new window which represent the RoadrunnerImporter. Use the plugin in like in the youtube video in step 4. After this you have your created road inside your own UE4 Level.
9. Now you have to search for the OpenDriveActor and drop it in the map. This process will generate the Routeplanner. In my case the Routeplanner lines spawned not directly on my street. I dont know if this is normal or not. I just moved my street mesh till it matched with the routeplanner lines.
10. The i searched for the CarlaMapGenartor Actor and droped it also on the map.
My CircleRoad had the Z Value 0 so i just changed the CarlaMapGenerator Z Value also to Z. After this step this small roadnetwork appears right next to my circle.
Then I pressed the "Trigger Road Map Generator" inside the CarlaMapGenerator (till know i dont know if we have to do this or not)
11. The last step was placing the "Vehicle Spawn Point" which you also can find like the other actors in the actor browser ( I dont know the right name for that)
Like you can see in the pictur i placed it direkt above the Routeplanner line. @XGodina Maybe this was your problem.

I hope someone will help this small instruction.

Cheers.

XGodina, 2018-12-10:
Congratulations @Atlunch !!!
I know the process is a bit mess. We are now working to make the process easier. This week we put new documentation about that. Can you see the semantic segmentation @Atlunch ? It would have to be as easy as moving the meshes in their corresponding folders. All information is welcome.

Thanks @Atlunch

Atlunch, 2018-12-12:
@XGodina 
That would be wonderful I'm looking forward for the new documentation.
Will the new documentation also include more about the new API?
With the manual_control.py I can use the semantic segmentation.

rmurrish, 2018-12-12:
@Atlunch 
Thanks for the detailed instructions, I'm struggling through this as well until the new docs are available.
In step 8/9, how did you search the OpenDriveActor and CarlaMapGenerator?

Atlunch, 2018-12-12:
@rmurrish 
On the left side you will find both in this "actor" browser. I don't know the real name :D

[image]

XGodina, 2019-01-23:
Vector Zero has a tutorial https://www.youtube.com/watch?v=OcbYuTFAo8o&t=125s. Maybe help you

geekhut, 2019-01-24:
@XGodina After enabling Roadrunner plugin in Unreal Engine, restart the project, Missing Plugin error box appears "Binaries for the 'RoadRunnerImporter' plugin are missing or incompatible with the current engine version. Would you like to disable it? You will no longer be able to open assets that were created with it"

I'm so done.

geekhut, 2019-02-11:
And where is OpenDriveActor? I don't see it anywhere..

XGodina, 2019-02-11:
Open Unreal and find it in menu Modes.

[image]

lijecaru, 2019-02-22:
> * Now you have to search for the OpenDriveActor and drop it in the map. This process will generate the Routeplanner Lines. In my case the Routeplanner lines spawned not directly on my street. I dont know if this is normal or not. I just moved my street mesh till it matched with the routeplanner lines.
> *

I tried following your tutorial, but when I drop the OpenDriveActor in the map it appears as an icon. Does that mean the roadrunner plugin is not working? I enabled the plugin without problems, but I don't see the roads generation so I suspect it is not working.

Atlunch, 2019-02-22:
If you have the opendrive actor marked, you have to click  inside the opendrive actor settings on the right side „generate road“ or something like that.
Thats new! Since last release.

lijecaru, 2019-02-22:
> If you have the opendrive actor marked, you have to click inside the opendrive actor settings on the right side „generate road“ or something like that. Thats new! Since last release.

I just found that out! Unfortunately, when I click that setting, my Unreal Project crashes and closes. Any ideas? Suggestions?

slyer0009, 2019-02-26:
@lijecaru have you tried rotating your map by 180° around the y axis (like it's mentioned in the instructions)?

lijecaru, 2019-02-28:
Thanks @slyer0009 that was the problem. Now it's working fine.

lijecaru, 2019-02-28:
> Thanks @slyer0009 that was the problem. Now it's working fine, but I have another question: how do I find the spawn points? I see that in my Unreal Project the spawn points are enumerated, but when I use the same number in my script to spawn a vehicle in a certain place, it appears in a totally different place. I tried using the number before/after thinking that it could be an indexation problem, but it didn't work.

I only found one thread related to this issue: [#1135](https://github.com/carla-simulator/carla/issues/1135)

zhang-xinxin, 2019-03-06:
@Atlunch Thanks for you sharing! I followed your tutorial, create a map using RoadRunner, import .fbx to UE4 Editor, place .xodr in Opendirve folder, place OpenDriverActor, VehicleSpawnPoints etc. All looks fine, and then I click "PLAY“ button in UE4 Editor. So I try to run the manual_control.py script, but failed. 
The error information as below:
------------------
  main()
  File "manual_control.py", line 842, in main
    game_loop(args)
  File "manual_control.py", line 768, in game_loop
    world = World(client.get_world(), hud, args.filter)
RuntimeError: std::bad_cast
--------------------

It seems that the client can't get the world information, right?
Anyone know that? What shall I do to fix it?

eloy96vg, 2019-04-23:
Hi,

I have trying to follow the steps indicated by @Atlunch , but i have problems importing the roadrunner plugin. I copy the plugin as indicated and I get the following errors:
[image]
[image]

After that, i delete it and reopen the .uproject but the plugin doesn't appear. 
In the tutorial of Carla's website nothing is said about the plugin and you can not drag all the generated meshes to the generated level. I can not import in any of the two ways indicated. Hope someone can help me. 

Thanks in advance.

Derekabc, 2019-04-23:
@eloy96vg  have you figure out how to import roadrunner plugin?

Derekabc, 2019-04-24:
**Hi, @Atlunch  Where can I find the "generate road“(I am using 0.9.5, I cannot find a choice like that).  I have the same issue faced by @lijecaru,  dropped the OpenDriveActor in the map it appears as an icon rather than the planned road.  BTW. @lijecaru , have you figured out how to solve this problem?**

eloy96vg, 2019-04-24:
Thanks @Derekabc, i have already installed the plugin, but when i create the mew level and import the fbx file (with the same name) i get a lot of staticMeshes but the map is not created as seen in the image. Would you know how to import the file correctly to create the map.

[image]

Thank you.

Derekabc, 2019-04-24:
@eloy96vg I import the fbx file into the UE, and drag and drop all the things to the created level.  I just created a simple road from RoadRunner, which you can see like this.
[image]
I deleted the floor by myself(I am not sure I am correct or not).

eloy96vg, 2019-04-25:
@Derekabc the problem is that i can't drag and drop into the level. Do you use a special configuration in Unreal Engine to do this?

ShanshanYOU, 2019-05-15:
> Finally I made it!!!!!!!!!!!!!!!!!!!!!!!!!
> 
> My System:
> Processor: i7-4790k
> RAM: 16 GB
> GPU: Nvidia GTX 980 Ti (Driver 384.130)
> OS: Ubuntu 16.04 (I had to switch from 18 to 16 because Roadrunner is actually only running on 16)
> 
> [image]
> [image]
> [image]
> 
> My steps creating my own map using Roadrunner and Carla 0.9.1.
> 
>     1. I downloaded the carla 0.9.1 source and built it like in the docs its mentioned.
>        Furthermore I built also the PythonAPI with make PythonAPI to tried the server and client connection.
> 
>     2. Like my example I created just a circle road in roadrunner.
> 
>     3. Export the FBX and OpenDrive using File -> Export ... you have to export both separatly.
> 
>     4. Import the Roadrunner Plugin into the CarlaUE4 Project File following this tutorial.
>        https://www.youtube.com/watch?v=OcbYuTFAo8o
>        You will find the plugin folder (on Ubuntu) in /usr/bin/VectorZero/Tools/Unreal/Plugins.
>        IMPORTANT: You have to follow these steps.
> 
> 
>     * Copy the Roadrunner Plugin into the Folder /home/yourname/carla91s/Unreak/CarlaUE4/Plugins.
> 
>     * Try to start the CarlaUE4.uproject..... it will fail.
> 
>     * Delete the Roadrunner Plugin
> 
>     * Try to start the CarlaUE4 again. It will start.
> 
>     * Open the Plugin Settings Edtit->Plugings, if there is a folder named Other open it.
> 
>     * Click Enable Roadrunner and restart.
> 
>     * If you cant see the Roadrunner Plugin, repeat the steps
>       Doing this you can install the Plugin into your project.
> 
> 
>     1. Now copy the exported OpenDrive xodr file into your carla project folder. Not with the Unreal Editor. The Unreal Editor cant read OpenDrive files xodr.
>        /home/yourname/carla91s/Unreal/CarlaUE4/Content/Carla/Maps/OpenDrive.
>        Name the File like @XGodina mentioned like you will name your Level later in the Editor.
> 
>     2. Open the CarlaUE4 Project with make launch-only.
>        7.Now generate a new Level with File -> New Level inside the Maps and name it like the OpenDrive file.
> 
>     3. Import the FBX file into a separatly folder in the content folder. Doing this it will open a new window which represent the RoadrunnerImporter. Use the plugin in like in the youtube video in step 4. After this you have your created road inside your own UE4 Level.
> 
>     4. Now you have to search for the OpenDriveActor and drop it in the map. This process will generate the Routeplanner Lines. In my case the Routeplanner lines spawned not directly on my street. I dont know if this is normal or not. I just moved my street mesh till it matched with the routeplanner lines.
> 
>     5. Then I searched for the CarlaMapGenartor Actor and dropped it also on the map.
>        My CircleRoad had the Z Value 0 so I just changed the CarlaMapGenerator Z Value also to Z. After this step this small roadnetwork appears right next to my circle, which you can see in one of the images.
>        Don't press the "Trigger Road Map Generator" inside the CarlaMapGenerator.
> 
>     6. The last step was placing the "Vehicle Spawn Point" which you also can find like the other actors in the actor browser on the left site inside the editor (I dont know the right name for that)
>        Like you can see in the image I placed it directly above the Routeplanner line. @XGodina Maybe this is your problem.
>        [image]
> 
>     7. Press play, start a python client and enjoy.
> 
> 
> I hope someone will help this small instruction.
> 
> Cheers.

Hello @Atlunch @XGodina,

Perfect ! Really appreciate for your clear documentation ! I've tried literally the same steps as you did but I didn't go with success. I'd like to ask a tiny question for these two steps that you've mentioned :+1:  "Open the Plugin Settings Edtit->Plugings, if there is a folder named Other open it."
"Click Enable Roadrunner and restart."

Did you re-set those plugin settings in UE Editor ? I didn't really find them though.

Look forward to hearing from you soon :+1: 

Cheers !

ShanshanYOU, 2019-05-15:
> Open Unreal and find it in menu Modes.
> 
> [image]

Hello @XGodina,

OpenDrive Actor is not in my list :( Any suggestions will be welcomed ! 

Cheers,

ShanshanYOU, 2019-05-15:
>     4\. OpenDriveActor

Hi, 

I'm wondering did you have the same problem that OpenDriveActor & CarlaMapGenerator do not exist in the list of modes ?

Cheers

ShanshanYOU, 2019-05-16:
> Finally I made it!!!!!!!!!!!!!!!!!!!!!!!!!
>
> My System:
> Processor: i7-4790k
> RAM: 16 GB
> GPU: Nvidia GTX 980 Ti (Driver 384.130)
> OS: Ubuntu 16.04 (I had to switch from 18 to 16 because Roadrunner is [...]

Hello @Atlunch & @XGodina ,

I've tried until the last step of your documentation. All went well unless the "RoutePlanner". It was not generated automatically by opendrive file. BTW that all directions & junctions are defined within RoadRunner and exported successfully. 

Could you please add some details about how to launch the agent with the Editor ? I've followed this [issue_1296](https://github.com/carla-simulator/carla/issues/1296), but I couldn't launch python script.

Cheers !

Dhairyashil-kpit, 2019-05-21:
@ShanshanYOU 
In you 4 th point : " Now you have to search for the OpenDriveActor and drop it in the map. This process will generate the Routeplanner Lines. In my case the Routeplanner lines spawned not directly on my street. I dont know if this is normal or not. I just moved my street mesh till it matched with the routeplanner lines. "

When I dropped OpenDriveActor on the map, it did not generate Routeplanner Lines. 
Though after following all further steps I clicked Play Button which gave me a black viewport. So I clicked Stop Button, which gave me following error log : 

> Blueprint Runtime Error: "Accessed None trying to read property StoredDirectionalLightActor". Blueprint:  BP_Weather Function:  Light Intensity Graph:  LightIntensity Node:  Set Intensity
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Light Intensity Graph:  LightIntensity Node:  Set Intensity
Blueprint Runtime Error: "Accessed None trying to read property StoredSkyLight". Blueprint:  BP_Weather Function:  Light Intensity Graph:  LightIntensity Node:  Set Intensity
Blueprint Runtime Error: "Accessed None trying to read property StoredDirectionalLightActor". Blueprint:  BP_Weather Function:  Orient Light Graph:  OrientLight Node:  SetActorRotation
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Colors determined by sun position
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Cloud opacity
Blueprint Runtime Error: "Attempted to assign to None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Cloud opacity
Blueprint Runtime Error: "Accessed None trying to read property StoredSkyLight". Blueprint:  BP_Weather Function:  Light Color Graph:  LightColor Node:  Set Light Color
Blueprint Runtime Error: "Accessed None trying to read property StoredDirectionalLightActor". Blueprint:  BP_Weather Function:  Light Color Graph:  LightColor Node:  Set Light Color
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Light Color Graph:  LightColor Node:  Set Light Color
Blueprint Runtime Error: "Accessed None trying to read property StoredDirectionalLightActor". Blueprint:  BP_Weather Function:  Light Color Graph:  LightColor Node:  Set Enable Light Shaft Bloom
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Light Color Graph:  LightColor Node:  Set Enable Light Shaft Bloom
[... cut ...]
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Fog Graph:  UpdateFog Node:  Set Fog Multiplier
Blueprint Runtime Error: "Accessed None trying to read property StoredAtmosfericFog". Blueprint:  BP_Weather Function:  Update Fog Graph:  UpdateFog Node:  Set Sun Multiplier
Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Fog Graph:  UpdateFog Node:  Set Sun Multiplier
Play in editor start time for /Game/Carla/Maps/UEDPIE_0_circle 2.414

Please help me with this, your help will be highly appreciated !

ShanshanYOU, 2019-05-21:
> [image]
> @ShanshanYOU
> In you 4 th point : " Now you have to search for the OpenDriveActor and drop it in the map. This process will generate the Routeplanner Lines. In my case the Routeplanner lines spawned not directly on my street. I dont know if this is normal or not. I just moved my street mesh till it matched with the routeplanner lines. "
> 
> When I dropped OpenDriveActor on the map, it did not generate Routeplanner Lines.
> Though after following all further steps I clicked Play Button it gives me a error log as shown in attached screenshot and then the viewport becomes black. So I clicked Stop Button, which gave me following error log :
> 
> > Blueprint Runtime Error: "Accessed None trying to read property StoredDirectionalLightActor". Blueprint:  BP_Weather Function:  Light Intensity Graph:  LightIntensity Node:  Set Intensity
> > Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Light Intensity Graph:  LightIntensity Node:  Set Intensity
> > Blueprint Runtime Error: "Accessed None trying to read property StoredSkyLight". Blueprint:  BP_Weather Function:  Light Intensity Graph:  LightIntensity Node:  Set Intensity
> > Blueprint Runtime Error: "Accessed None trying to read property StoredDirectionalLightActor". Blueprint:  BP_Weather Function:  Orient Light Graph:  OrientLight Node:  SetActorRotation
> > Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Colors determined by sun position
> > Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
> > Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
> > Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
> > Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
> > Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
> > Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
> > Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
> > Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
> > Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
> > Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
> > Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
> > Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Scalar Parameter Value
> > Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
> > Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
> > Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
> > Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
> > Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
> > Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Vector Parameter Value
> > Blueprint Runtime Error: "Accessed None trying to read property StoredSkySphere". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Cloud opacity
> > Blueprint Runtime Error: "Attempted to assign to None". Blueprint:  BP_Weather Function:  Update Sky Graph:  UpdateSky Node:  Set Cloud opacity
> > Blueprint Runtime Error: "Accessed None trying to read property StoredSkyLight". Blueprint:  BP_Weather Function:  Light Color Graph:  LightColor Node:  Set Light Color
> > Blueprint Runtime Error: "Accessed None trying to read property StoredDirectionalLightActor". Blueprint:  BP_Weather Function:  Light Color Graph:  LightColor Node:  Set Light Color
> > Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Light Color Graph:  LightColor Node:  Set Light Color
[... cut ...]
> > Blueprint Runtime Error: "Accessed None trying to read property StoredAtmosfericFog". Blueprint:  BP_Weather Function:  Update Fog Graph:  UpdateFog Node:  Set Sun Multiplier
> > Blueprint Runtime Error: "Accessed None". Blueprint:  BP_Weather Function:  Update Fog Graph:  UpdateFog Node:  Set Sun Multiplier
> > Play in editor start time for /Game/Carla/Maps/UEDPIE_0_circle 2.414
> 
> Please help me with this, your help will be highly appreciated !

@Dhairyashil-kpit Wrong person bro ! That might be @Atlunch who succeded in this task. I actually asked a similar question as u did : Nothing happened with OpenDriveActor.

Dhairyashil-kpit, 2019-05-22:
@Atlunch Please Help us regarding this : 

> [image]
> @ShanshanYOU
> In you 4 th point : " Now you have to search for the OpenDriveActor and drop it in the map. This process will generate the Routeplanner Lines. In my case the Routeplanner lines [...]

Dhairyashil-kpit, 2019-05-24:
Hi,
Kindly refer : https://github.com/carla-simulator/carla/issues/1662#issuecomment-495484746 
For steps to import map in CARLA

closed 2019-07-30 by stale[bot]

NebulousNeo, 2019-09-05:
I found a solution, it was the fault of the .egg file in the folder there wasn't the right .egg file version (carla-0.9.5-py2.7-linux-x86_64.egg instead of should be carla-0.9.6-py2.7-linux-x86_64.egg). This problem occur because i had two carla folder the first one was the old carla version(0.9.5) and the new one(0.9.6 build from source). So i copied from folder /home/xxx/Test/carla/PythonAPI/carla/dist/ (folder of 0.9.6) the egg files to /home/xxx/yyy/zzz/Carla/CARLA_0.9.5/PythonAPI/carla/dist (this is 0.9.5 where i also execute my pythonAPI). after that it worked and i could conntect to the carla world.