carla-simulator/carla #1016: [0.9.1] [Linux] Build from source, Server has very low FPS

Atlunch, 2018-12-05:
Hello everyone,

I just build a fresh 0.9.1 carla build on ubuntu 18.04 from source.
My Hardware:
Processor: i7-4790k
RAM: 16 GB
GPU: Nvidia GTX 980 Ti (Driver 410)

I opened the UE4 Editor and start the sever by pressing play.
After that I connected the python client "manual_control.py"
Iam getting a connection and everythin is working.

But the FPS of the server is with 2 - 3 FPS really low. The Client FPS is around 60 FPS.

While using the pre compiled 0.9.1 version I'm getting much better results.
Server around 20FPS and Client also around 60 FPS.

What I'm doing wrong?

yyuuliang, 2018-12-06:
I have the same problem if I directly use UE4 editor.

Atlunch, 2018-12-06:
@yyuuliang
You are saying "if i direclty use UE4 editor"....does that mean that usually you have another way?
It is possible to create your own map with the "build from source version" and use it with the precompiled version?
Or is it also possible to create with the "build from source version" a server programm like in the compiled version which you start with ./CarlaUE4.sh ?

yyuuliang, 2018-12-06:
@Atlunch 
First thing I noticed is that UE editor didn't work well on my Ubuntu 16. So I switch to widnows 10.
If you modify your maps and want to test it, check this #1017 
If you want to test your map chagnes with the PythonAPI, first run "make clean", this will remove all the saved maps, then "make launch", this will create the binaries with the new map. On windows, after that, I can find CarlaUE4.exe here: \Unreal\CarlaUE4\Binaries\Win64>CarlaUE4.exe. Then you can start the server as in the document. And then start the PythonAPI to connect it to the server.

closed 2019-02-16 by stale[bot]