carla-simulator/carla #1017: Is there an easy way to test the map changes?

yyuuliang, 2018-12-06:
Hi
I am recently developing my own map on windows 10 using UE 4.19.2 and Carla 0.9.0. I use "make launch" to build the new CarlaUE4.exe. I found that this cmd can't load the new changes in the map. So now every time I want to test the map, I have to "make clean && launch".
I also tried to "Play" the map using the UE editor, then start the PythonAPI to connect to it. However, I found that Cara becomes extremely slow. 
Is there an easier way to test the map?

Thanks

TheNihilisticRobot, 2018-12-06:
Hi @yyuuliang 

The easiest way to test a map you are modifying is to start a game using the old gamemode (CarlaGameMode). 

In the Toolbar, open the "Blueprints" menu, then, under project settings, open the "GameMode: Edit..." menu, then "Select GameModeBase Class" and choose CarlaGameMode. If you play from the editor with this gamemode you'll automatically spawn a vehicle you can drive around.

About the second question, if you've copied exactly the line you use to open the game you'd be missing the file extension of the map, try using: `./CarlaUE4.exe /Game/Maps/Town04.umap` instead.

yyuuliang, 2018-12-06:
@TheNihilisticRobot thanks! That really helps!

yyuuliang, 2018-12-11:
hi @TheNihilisticRobot can you list the differences between those modes? I changed the gamemode to CarlaGameMode, it worked. But if I compiled the binaries using this mode, the pythonAPI couldn't connect to the server. I also tried to use "CarlaUE4.exe -carla-server", this didn't work, either. So now I don't know how to make this work...

What mode is the default for the carla-server?  How can I make my compiled binary run as the server by default?

TheNihilisticRobot, 2018-12-11:
I'm sorry, that's expected to happen. Forgot to tell you to change the gamemode again. CarlaGamemode is the old mode, from before the new API. Before compiling you'll need to change the gamemode for the client to work. Same procedure than before but selecting the "NewCarlaGameMode".

yyuuliang, 2018-12-11:
@TheNihilisticRobot  Thank you so much for the speedy reply! It works again ;)

closed 2019-02-17 by stale[bot]