carla-simulator/carla #1019: Draw_Waypoints error Carla 0.9.1

aroongta, 2018-12-06:


Sakrust, 2018-12-07:
Because there is no function 'draw_waypoints' in the API, you have to define this function by yourself.

nsubiron, 2018-12-10:
This is the function we used
```python
def draw_waypoints(world, waypoint, depth=6):
    if depth < 0:
        return
    for w in waypoint.next(4.0):
        t = w.transform
        begin = t.location + carla.Location(z=0.5)
        angle = math.radians(t.rotation.yaw)
        end = begin + carla.Location(x=math.cos(angle), y=math.sin(angle))
        world.debug.draw_arrow(begin, end, arrow_size=0.3, life_time=1.0)
        draw_waypoints(world, w, depth - 1)
```

aroongta, 2018-12-10:
Thank you so much @Sakrust and @nsubiron

closed 2018-12-11 by marcgpuig

shushu-nantong, 2019-08-28:
@nsubiron 
Would you please explain the code about "begin = t.location + carla.Location(z=0.5)" since t.location already outputs the location.
In other words, what does "carla.Location" stand for?

Thank you very much.