carla-simulator/carla #1020: [0.9.1] [Linux] Build from source, ./CarlaUE4.sh Failed

Atlunch, 2018-12-07:
Hello everyone,

I build a fresh 0.9.1 carla build from source.
I'm using Ubuntu 18.04 and the nvidia driver-410.
I can run the precomiled 0.9.1 build and I can UE4 Editor without any problems.
Also I'm able to start the server from the "build from source" version in my UE4 Editor.
Now I would like to start the "source" version like the precompiled without using UE4 Edior.
For this I used the CarlaUE4.sh from the precompiled version and changed only the path for the CarleUE4 executable like you can see in the image.
But tying to run this gives me following error....
[image]
 
I found a few issues related to #757.
But I'm able to use the UE4 Editor.
Or is this the wrong way to start the source version?

fabianoboril, 2018-12-09:
Hi,
when you build from source, you should run "make package". This will create a CARLA package similar to the pre-compiled one. In the "Dist" folder you should be able to find the new CarlaUE4.sh that should work with your build.

Atlunch, 2018-12-09:
@fabianoboril  Thank you for your information.

In my carla/PythonAPI/dist I found only two zip.files which arent include a .sh file.
I ran make package again. It seams like it works but it was really fast and at the end i get this error:

..........
BuildPythonAPI.sh: Success!
fatal: Not a git repository (or any of the parent directories): .git
Util/BuildTools/Linux.mk:13: recipe for target 'package' failed
make: *** [package] Error 128

What does that mean?

closed 2019-02-14 by stale[bot]

ravishk1, 2019-03-14:
Hello, I am getting the same error in **"make package"** command.

:::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::
AutomationTool exiting with ExitCode=25 (Error_UnknownCookFailure)
RunUAT ERROR: AutomationTool was unable to run successfully.
Util/BuildTools/Linux.mk:13: recipe for target 'package' failed
make: *** [package] Error 25
::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::

How to solve this problem ??