carla-simulator/carla #1021: no effect with FPS and benchmark flag

tanglrHello, 2018-12-08:
Hi,
I started the carla server in this way:
    ./CarlaUE4.sh -benchmark -fps=10 -windowed

And then run manual_control.py. I'm sure manual_control.py is connecting to the carla server started just now in that I can see the same vehicle in the same position in pygame and server scene. But in the hud, the server FPS is always around 30 despite the value of fps flag, which is almost the same as when I started the carla server without benchmark and fps flags. I'm very confused about it. Appreciate any suggestion. Thanks.

nsubiron, 2018-12-10:
Hi @tanglrHello,

The FPS shown in the "manual_control.py" is measured in real-time, with a timer, just to test the performance of the server. What you may want to see is the simulation-time, e.g.
```diff
--- a/PythonAPI/manual_control.py
+++ b/PythonAPI/manual_control.py
@@ -307,7 +307,7 @@ class HUD(object):
 
     def on_world_tick(self, timestamp):
         self._server_clock.tick()
-        self.server_fps = self._server_clock.get_fps()
+        self.server_fps = 1.0 / timestamp.delta_seconds
         self.frame_number = timestamp.frame_count
         self.simulation_time = timestamp.elapsed_seconds
```

tanglrHello, 2018-12-10:
It works! Thank you so much.

tanglrHello, 2018-12-17:
In my understanding, if I set the fps of the simulator to 10, OnTick() will be executed 10 times per second. If I spawn a sensor in the carla world and add a callback function for the sensor, the callback function will also be executed 10 times per second at the same pace with OnTick(). But when I add a print time command in _parse_image() of manual_control.py, it shows the print lines are around 30 times in a single second, which means the callback function are executed 30 times per second. And as I modified the code in the way you suggested, server fps in hud is around 10.
I'd like to know what is wrong with my comprehension. Expecting your reply. Thank you!

nsubiron, 2018-12-17:
`OnTick()` callbacks are called 10 times per _simulation-second_, but it's actually called as fast as your hardware is able to compute. Meaning, every object in the simulation sees steps of 0.1ms, but you will observe the simulation like moving fast-forward. On the opposite, if you set an FPS of say 500, the vehicles are updated 500 times per _simulation-second_, but you'll observe the simulation in slow motion.

In other words, if you run with `-benchmark -fps=60`, save the images, and play them at 60 fps the video will look real-time.

tanglrHello, 2018-12-20:
It's clear to me now. Thanks for the reply!

closed 2018-12-20 by tanglrHello