carla-simulator/carla #1022: How can  I set the starting position?

youssefkhalil96, 2018-12-08:


fabianoboril, 2018-12-09:
Hi,
could you please provide a few details? Where do you want to set the starting position, and of what (ego vehicle?)?

youssefkhalil96, 2018-12-09:
Actually I want to make lane keeping control and I want to know the position of the vehicle within the two lanes so is there any way to do this thank you so much

youssefkhalil96, 2018-12-10:
Hi,
thank you so much for your attention. The attached vehicle is the one I am simulating on. Actually I am doing lane keeping control so I need to know is there any way to know the position of the vehicle from the center of the lane and also I need to know the width if the road thank you so much.

yyuuliang, 2018-12-12:
use the segmentation camera to get the lane info, there should be a way to transform the lane location into the world coordinate, then you can use this data to control the vehicle. Your vehicle's controller should have a method to do the lane keeping after receiving the lane coordinate.

closed 2019-03-01 by stale[bot]