carla-simulator/carla #1023: apply_control() is sticky

praveen-palanisamy, 2018-12-09:
The `apply_control(...)` method is "sticky" meaning, at every tick the previous control values are used by the simulator unless the values are overridden with another apply_control call per vehicle on every tick. Is this a correct observation?

 If so, this behavior may not be desirable even in async mode (the only mode as of 0.9.1). The ideal expected outcome is coasting/loss-of-control behavior when there are no new control commands. Is this possible with the existing API?

**CARLA version**: 0.9.1

nsubiron, 2018-12-12:
Resetting the control on each tick would require sending a control message every tick, that will increase significantly the amount of messages sent specially when controlling a lot of vehicles. Right now the control is cached and only sent if changed. Also controlling the vehicle might be affected, specially in things like manual control, if a message arrives too late the vehicle would reset the steering to zero just one frame.

praveen-palanisamy, 2019-02-11:
@nsubiron : In our follow-up discussion, we more or less agreed that there should be an appropriate/configurable command (with coast behavior/loss-of-control behavior/default-behavior) that is sent to the server by the API if there was no command sent by the client for a tick rather than the API re-using the old command sent by the client.
Is this still a work in progress under the `IdleControl` Epic or was something else decided?

nsubiron, 2019-02-11:
Hi @praveen-palanisamy,

Sorry I forgot to tag this issue as backlog and the bot almost closes it. The plan is to add some flag to the vehicles, so we can switch the "non-sticky" behaviour on or off per vehicle.

nsubiron, 2019-03-01:
This changes entered master in #1322.

closed 2019-03-01 by nsubiron