carla-simulator/carla #1029: How to create a new sensor (IMU) without blueprint?

YashBansod, 2018-12-11:
Hello,

I want to create a new sensor type **IMU** in my Client that publishes Ego Vehicle Information every simulation cycle.

Currently you would have to get this information by requesting the vehicle information in the client's game loop,
    - `vehicle.get_transform()`
    - `vehicle.get_velocity()`
    - `vehicle.get_acceleration()`
    - `vehicle.get_vehicle_control()`

However, since the Server and Client run asynchronously, it becomes a bit challenging to get this information at a constant cyclic rate.

Since the `Carla.Sensor` Class is a child of `Carla.Actor` Class, I was wondering if there was a way to create a sensor using the vehicle actor.
**or**
If there was a way to create a new sensor given a vehicle's blueprint.

This would enable the implementation of the `sensor.listen(callback)` method. Which could be used to write the ego vehicle information at the simulation rate.

If there's any means of implementing a sensor like this, please help me in its implementation.

nsubiron, 2018-12-11:
The easiest way of doing this is registering a callback on every tick in Python, so you don't need to modify any of the C++ code
```py
world.on_tick(lambda timestamp: do_stuff())
```
I think with this you have all the necessary components for implementing a sensor in pure Python.

paulaurel, 2018-12-14:
@nsubiron, @YashBansod how do I get the `vehicle.get_acceleration()`. The current `game.py` only provides ` measurements.player_measurements.acceleration`, which is the acceleration measured in the global coordinate frame, which is inertial. An IMU provides measurements in the vehicle's local non-inertial coordinate frame. I would love to have an IMU sensor that provides ego acceleration measurements with respect to the vehicle's center of mass.

YashBansod, 2018-12-14:
@nsubiron is it okay to have multiple `world.on_tick(lambda timestamp: do_stuff())` statements in the definition of **World**.
Presently `world.on_tick(hud.on_world_tick)` is already there in the definition of **World** Class: 
https://github.com/carla-simulator/carla/blob/77de6a958cb634b439d8139f283078866a6e55d5/PythonAPI/manual_control.py#L138-L150
If I write another `world.on_tick(lambda timestamp: do_stuff())` below, will it override the previous one or will both the methods be called every tick.

YashBansod, 2018-12-14:
@paulaurel This is in reference to the 0.9.1 API. This too gives the acceleration in the Global Coordinate Frame (I suppose). However, since you are also getting the Ego Vehicle's Transformation in the Global Coordinate, I think the transformation of measurements to Ego Frame should be fairly straight forward.

nsubiron, 2018-12-14:
@YashBansod `world.on_tick(callback)` registers a new callback, you can register as many as you want and all of them will be called on each tick. On the other hand, calling `sensor.listen(callback)` overwrites any previous callback on the sensor, there is a difference in behaviour here.

paulaurel, 2018-12-18:
@YashBansod after spending some time thinking about the transformation from the global coordinate frame to the Ego Vehicle. I believe that that transformation is not that simple. For the pose the following is true right: 
`r_vehicle = Q_(rotation_matrix)r_global`
Yet, for the acceleration you need to differentiate both  the rotation matrix and `r_global` with respect to time. That isn't that easy.

closed 2019-01-07 by YashBansod