carla-simulator/carla #1032: transforms and frames with different fps

Unlingius, 2018-12-11:
Hello! How can i get transforms and frames with different fps?  For example, 30 frames per second, and 100 transforms per second.
Thank you all!

nsubiron, 2018-12-13:
Do you mean receiving in the client side the "tick" message (that contains the transforms) and the images from the cameras at different frame rates?

If so, that's not currently possible. However I would like to add at some point an attribute to the sensors for controlling their frame rate, e.g. render only every X seconds, that could be great for performance in some situations.

I doubt however I will have time to do this any time soon :(

Unlingius, 2018-12-14:
Thank you! It seems, at least i found a workaround. First, i run Carla with Low quality rendering setting with -benchmark -fps 100 and only one sensor of 320x240 and store track in file. Second, i disable physics simulation for cameras and make images offline according to stored track. For now it is enough for me.

closed 2019-02-19 by stale[bot]