carla-simulator/carla #1034: Error make package: rsync error

zhangliang3, 2018-12-12:
Hi, I confront an error when I run "make package":   

BUILD SUCCESSFUL
AutomationTool exiting with ExitCode=0 (Success)
Package.sh: Adding extra files to package.
LICENSE
CHANGELOG.md
release_readme.md
python_api.md
Release.Dockerfile
rsync: link_stat "/carla/PythonAPI/dist/*.egg" failed: No such file or directory (2)
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1183) [sender=3.1.1]
Util/BuildTools/Linux.mk:13: recipe for target 'package' failed
make: *** [package] Error 23

My environment is ubuntu 18.04, Python3. Could anyone help me? Thanks!

nsubiron, 2018-12-14:
Never seen this error, it seems `make PythonAPI` failed.

closed 2019-02-19 by stale[bot]

huangxueqin123, 2019-09-09:
I also have same issue. can you have solve it?