carla-simulator/carla #1037: [0.9.1] [Linux] Implementing new weather szenario "night", CarlaWeather.ini

Atlunch, 2018-12-12:
Hello everyone,

I would like to create a night szenario in Carla.

I tried to create a new weather setting inside "CarlaWeather.ini". But I couldn't select it while i was using the manual_control.py.

Is this the right way or is there a better alternative?

And If this is the right way, does anyone have already the the right weather setting for a night szenario?

Atlunch, 2018-12-14:
Great! Thank you very much.
While changing a few settings inside the editor it is possible to get a real night feeling.

Unlingius, 2019-01-28:
@Atlunch  It looks really impressive! What settings did you change? How to attach light to car? I'm beginner in UE

closed 2019-04-05 by stale[bot]

YDSh510, 2019-05-06:
@Atlunch - Which settings did you change to get above night effect in UE4 editor?

AlanNaoto, 2019-11-20:
> Great! Thank you very much.
> While changing a few light settings inside the editor it is possible to get a real night feeling.
> [image]

Hey! I'm really interested on this feature as well; would you mind sharing the assets/how to do that please?

rrrokhtar, 2021-05-01:
@rikardomarenzzi  Can you mention the settings please