carla-simulator/carla #1039: How can I do the autopilot with specific starting point and destination to get sensor data and pilot data in 0.8.2?

zxbnjust, 2018-12-13:
Since the Client_example.py has no destination,and if I add "control = measurements.player_measurements.autopilot_control" in agent,the car will just go straight.So how can I use the autopilot to get data?
Thank you for any answer!

nsubiron, 2018-12-13:
Hi @zxbnjust,

The autopilot decides at random at every intersection, we don't have a way of setting specific destinations.

zxbnjust, 2018-12-14:
> Hi @zxbnjust,
> 
> The autopilot decides at random at every intersection, we don't have a way of setting specific destinations.
So how can I get the AI drive information to train my nets?

nsubiron, 2018-12-14:
`measurements.player_measurements.autopilot_control` gives you that, this is the control the autopilot would give this frame.

zxbnjust, 2018-12-14:
Since the autopilot decides at random at every intersection,I don't know the usage of autopilot.I hope set the destination for AI pilot.Is there any way to achieve that?

nsubiron, 2018-12-14:
No, that's not currently possible. You can retrieve the current autopilot control but not where is it going.

zxbnjust, 2018-12-15:
https://github.com/carla-simulator/carla/issues/288#issuecomment-374700793
What do you think about this? I think the autopilot.py exist before 0.8.1,but in 0.8.2,I can't find the file.Maybe I'm not very clear about my needs,but I have the same question with the link.Thank you for your trouble.

nsubiron, 2018-12-15:
Ah ok I see, that's not our autopilot (what you get with player_measurements), that seems to be a client-side autopilot that @felipecode was writing on a separate branch of Carla. I don't know how it works and if it's compatible with 0.8.2. Maybe @felipecode can help us with that?

jingzhao3200, 2019-02-06:
> Ah ok I see, that's not our autopilot (what you get with player_measurements), that seems to be a client-side autopilot that @felipecode was writing on a separate branch of Carla. I don't know how it works and if it's compatible with 0.8.2. Maybe @felipecode can help us with that?

> Hi @zxbnjust,
> 
> The autopilot decides at random at every intersection, we don't have a way of setting specific destinations.

Does it still have no way to set destinations in Carla 0.9.2 version? I'd like to use autopilot mode to set some certain start and end points, but I did not fide where the definition of function **world.vehicle.set_autopilot**(line 228 in automatic_control.py) source code?

closed 2019-04-14 by stale[bot]

Derekabc, 2019-05-06:
@jingzhao3200 hi, do you have any ideas about set destinations?  I know we can try to set destination at basic_agent.py. But i failed, the can cannot achieve the setted destination.

jingzhao3200, 2019-05-06:
> @jingzhao3200 hi, do you have any ideas about set destinations? I know we can try to set destination at basic_agent.py. But i failed, the can cannot achieve the setted destination.

also found not works for set_destination. But recently I did not work on the latest version of carla, if I have found solutions, will let you know.

Deepak3994, 2019-05-09:
Hi,

Even i am facing the same issue. The path is not getting generated to the desired destination. I am using carla 0.9.3.

Kindly help..

bijayaniSMishra, 2021-04-21:
Hi.. I am using 9.10, Is there a way to set destination for Carla in autopilot mode?

zenyanagata, 2022-07-06:
Same here. Is there any way to set a destination, like waypoints, for autopilot mode?

JamesLuyangJiang, 2023-06-06:
I think starting from 0.9.12, there is a `set_destination` method from CARLA API, which allows you to choose a location in the map for the agent to travel to in shortest path. Here is the relevant documentation: [Link to CARLA 0.9.12 Documentation](https://carla.readthedocs.io/en/0.9.12/adv_agents/#:~:text=set_destination(end_location%2C%20start_location%3DNone,use%20the%20current%20agent%20location.)

Hope this helps!

Micahmichael03, 2024-08-09:
> Hi,
> 
> Even i am facing the same issue. The path is not getting generated to the desired destination. I am using carla 0.9.3.
> 
> Kindly help..

have you found a way?