carla-simulator/carla #1040: [0.9.1][Linux] no raindrops in 4AxisWeatherSystem

saminkhan, 2018-12-13:
I've tried with compiled binary as well as the source version of 0.9.1.  In both cases, I am not seeing any raindrops when precipitation level is non-zero.  I have checked spectator camera as well as captured frames.  Puddle deposits are behaving correctly, with proper ripples, but there are no actual raindrops.  What could be causing this issue?  Am I forgetting to set a parameter somewhere?

TheNihilisticRobot, 2018-12-13:
Hi @saminkhan 

I don't think you're missing any step.
Spectator camera won't have rain particles at all, that's its expected behaviour.  Once you configure the precipitation above 0 you should see rain particles in front of any Scene Capture Camera you spawn in the world (Or that spawns with the vehicle).

If you have the source version there's a little test you could do. Spawn a Scene Capture Camera and a BP_Weather in the world, set the precipitation level to 100 and check if raindrops start to appear. Make sure that RealTime is checked in the Viewport options (The small arrow pointing down in the top left corner of the viewer)

[image]
 
Please, let me know if you still don't get rain particles and I'll check what could be causing it.

saminkhan, 2018-12-14:
Thanks for the reply @TheNihilisticRobot .  The test that you've mentioned works fine in the Editor, and I can now understand why spectator mode doesn't show raindrops.

But I'm still not seeing any raindrops in sensor.camera.rgb output images.  I've modified tutorial.py by following the suggestion on [issue #1037](https://github.com/carla-simulator/carla/issues/1037#issuecomment-446651358).  In output from sensor.camera.rgb, I see a depth blur effect associated with increased precipitation, but no sign of raindrops.  Precipitation is set to 80, and I check for 1000 frames.

TheNihilisticRobot, 2018-12-14:
I'm sorry I didn't think of this earlier but I think I know what could be causing the issue.

Have you setted the precipitation before spawning the vehicles? 
Right now our weather checks for cameras each time you run it. Vehicles don't check the weather on spawn to get the particles attached.
You could refresh the weather every time you spawn a vehicle or wait until all of them have spawned to set the weather. I'll note this to add to the documentation and to make the vehicle spawn with particles when is raining.

saminkhan, 2018-12-14:
Ah, makes sense!  I've moved set_weather call after spawning the camera, and now I see raindrops in the images.  Thanks again.

closed 2018-12-14 by saminkhan