carla-simulator/carla #1041: Python API: carla.Image depth convertion

danil-tolkachev, 2018-12-13:
Currently this code doesn't work
```python
carla_image.convert(carla.ColorConverter.Depth)
carla_image_data_array = numpy.ndarray(
   shape=(carla_image.height, carla_image.width, 1),
   dtype=numpy.float32, buffer=carla_image.raw_data)
```

Actually method `carla.Image.convert(carla.ColorConverter.Depth)` converts to RGBA format. It's may be useful (in `manual_contorl.py` for example) but it's not user may expected. Possible solution may be to change and extend existing `carla.ColorConverter` values:
```
Raw
Depth # to float32 format
DepthImage # to bgra format
LogarithmicDepth # to float32 format
LogarithmicDepthImage # to bgra format
CityScapesPalette
```

nsubiron, 2018-12-13:
Just to add some thoughts on why is done like this. The `image.convert` function calls the C++ `ImageConverter::ConvertInPlace`, this function is quite efficient because it avoids allocating data by overwriting the image buffer that we receive from the server, this is great cause the buffers we receive from the server are reused (i.e. when an image goes out of the scope its memory is reused for another image). The problem is that the pixel type is part of the image type, so changing the pixel type changes the type of the image and becomes a bit challenging converting an image in place. 

Also at it is now images are always BGRA which simplifies the API for the user in my opinion.

But yeah, in the case of the depth this sucks because you're actually loosing precision converting to uint8 grayscale.

Maybe we can add functions to create numpy arrays directly from the images?
```py
image.to_numpy_array(cc=Raw, format='bgra') 
image.to_numpy_array(cc=Depth, format='float32')
```

closed 2019-02-18 by stale[bot]

soulslicer, 2019-08-20:
Hi, can you provide the function you used to convert to get a regular depth map please? I still cannot figure it out