carla-simulator/carla #1042: Get multiple Cameras' data at the same time from the python client side

bigsheep2018, 2018-12-13:
As the title mentioned, is it possible to get multiple cameras' data at the same time on the python side?

E.g., get the depth camera and lidar camera's data at the same time?

Thanks in advance.

marcgpuig, 2018-12-13:
Hi @bigsheep2018,
Please, add in which Carla version are you working, as our issue creation protocol specifies.

bigsheep2018, 2018-12-13:
Hello @marcgpuig,
The Carla version is [0.9.1]
The title is modified.
So is it possible to get multiple sensors' data at the same time?

fabianoboril, 2018-12-13:
You can create multiple cameras. Then, each image comes with a frame number. Hence, you should be able to synchronize data coming from multiple cameras. Does this help you?

fabianoboril, 2018-12-14:
Hi,
yes kind of. You spawn multiple cameras. For each you register a callback to receive the image. Each image comes with a framenumber. So you receive a set of images for each camera. Then, you can correlate/fuse them using the framenumber (the framenumber should correspond to the timestamp of the image).

bigsheep2018, 2018-12-17:
@fabianoboril  Thanks. Close the issue.

closed 2018-12-17 by bigsheep2018

KouWeiBin, 2022-07-28:
> You can create multiple cameras. Then, each image comes with a frame number. Hence, you should be able to synchronize data coming from multiple cameras. Does this help you?

Hi, In think you are right, but how to sync the frame number of multiple cameras?

DaniCarias, 2024-03-12:
> > You can create multiple cameras. Then, each image comes with a frame number. Hence, you should be able to synchronize data coming from multiple cameras. Does this help you?
> 
> Hi, In think you are right, but how to sync the frame number of multiple cameras?

+1