carla-simulator/carla #1045: Traffic light detection :how to label the traffic lights in the image?

JackRen88, 2018-12-14:
Hello,everyone!
In the traffic light detection, I want to label the traffic lights in the image taken from Carla in order to train my detection model,I don't know if there is a way to get the annotation image in Carla.if so,Is it possible to provide pixel coordinates  at four points of bounding box labeled in the image and traffic lights status  which is green or red,orange?

I hope to get your help.

chowkamlee81, 2018-12-19:
@JackRen88 , did you got to any clue on the smae since iam also struggling

1453042287, 2018-12-28:
@JackRen88 @chowkamlee81 why not just use the image of semantic segmentation ?

closed 2019-03-05 by stale[bot]

micmarty, 2019-04-04:
Anything changed? How to easily automate traffic light dataset gathering process for object detection purposes (ground truth bounding box on camera rgb image)?

My plan was to use Tensorflow Object Detection API :)