carla-simulator/carla #1046: Carla server seems stuck for a while after I spawn a vehicle and add a lidar sensor

tanglrHello, 2018-12-14:
carla version: 0.9.1
OS: ubuntu 16.04
GPU: GEFORCE GTX 1080Ti

The problem can sometimes be reproduced in this way:
1. Start carla server without setting fps
2. Start a client program, which spawn a vehicle in carla server and attach a lidar sensor to it. And I set PointsPerSecond for this sensor to 500000.
3. The client will then immediately get the sensor data from lidar, and the sensor data will be computed some way. Then the client produce a new location for the vehicle, and set the new location to carla server continuously. We have another visualization program, in which we can see the car is moving with expectation. The car should accelerate and moving smoothly.
4. But sometimes ( not everytime ), the car in carla server scene seems to be stuck and will not move in the same way as it move in another visualization program. It doesn't move for seconds, or less a second. Then the SetTransform() request accumulated are executed now, the car moves very fast in carla server until reaching the position it should be now.
5. In the client, I added some log before calling SetTransform(), and the log are produced normally. So SetTransform() requests are not blocked in the client side.

This will happen more frequently with a sensor defined by ourselvs.

I'd like to know the reason for this phenomenon. It is due to the gpu is not powerful enough? But expect for the few seconds at the beginning, things are ok afterwards. The carla server is not lagging and I can move the view in it normally.

nsubiron, 2018-12-14:
The `set_transform`, `apply_control`, and these requests are sent asynchronously to the server (i.e. the client does not wait until the request has been fulfilled). At this point, if the server is too slow parsing requests, the requests will start accumulating on a queue. This is a known issue, and would require to modify the rpclib to fix it I'm afraid. It's possible that your sensor is adding more charge to the simulator so it has lower frame-rate then is able to parse requests less often (the simulator has a limited amount of time each tick to parse requests, currently 10ms).

The way I usually avoid this is synchronizing the client with the game ticks with `world.wait_for_tick()` function. This function blocks until a new tick is produced by the server, this way you make sure you send requests at the same frame-rate as the simulator.

tanglrHello, 2018-12-15:
Thank you for the explanation! But I still wonder why the accumulation of requests is especially obvious at the beginning. In my experiment, the accumulation usually disappear after a few seconds. And sometimes after a few minutes, the carla server seems getting completely stuck, and no requests of set_transform and apply_control are executed for a long time. But lidar sensor data can be received. If the simulator has 10ms to parse requests in each tick, the car should move somehow.

closed 2019-02-20 by stale[bot]