carla-simulator/carla #1050: Where to find dynamic specs of car ,  for controlling car  smoothly

jediofgever, 2018-12-15:
CARLA version; 8.0.2
Envoirement; UBUNTU 16.04 LTS

Hi,
Thank you for enabling such amazing simulator. Currently we are using CARLA to test our autopilot framework based on ROS. In one of the scenarios we are trying to control car on a highway. We can easily control the car in Gazebo, however in CARLA we need to manually change control gains to adjust controller to CARLA, and this breaks modularity of our software. I would like to ask where I can find specs of dynamics for car in CARLA? , in Gazebo it is easy to play with dynamics of car like by changing inertias , mass etc. 

I would like to make the dynamics of cars same at both simulator so that I don't need to adjust controller gains. 

anyone can uide me with this ?

Thank you very much

SpakMiguel, 2018-12-17:
If you open up the Unreal project you should be able to change mass, steering, etc. of vehicles and even creating your own ones.

jediofgever, 2018-12-17:
what do you mean by Unreal project ? , do you have link to it  ?

closed 2019-02-22 by stale[bot]