carla-simulator/carla #1051: [0.9.1] [Linux] Semantic Segmentation results are all black

xksteven, 2018-12-16:
When using the manual_control.py the semantic segmentation images captured with both raw and city scapes pallete both save images with all zeros and alpha value of 255.  I captured the images by "recording" them i.e pressing R.

I tested this with both python2.7 and python3.6 and both produced the same issue.

nsubiron, 2018-12-16:
Hi @xksteven,

Thanks for report! A few questions, cause I'm not able to reproduce this
* What's your OS version?
* Are you using our release binary or you compiled Carla yourself?
* Do you also see black images in the view of manual_control.py?

yyuuliang, 2019-01-26:
Did u check the image data in python? The image is black but they are not all 0. It should be something like (3,0,0), (4,0,0)... each number represents a segmentation class.

HeYDwane3, 2019-02-25:
> Did u check the image data in python? The image is black but they are not all 0. It should be something like (3,0,0), (4,0,0)... each number represents a segmentation class.

Can you tell me how to check the image data in python?

xksteven, 2019-02-27:
> Can you tell me how to check the image data in python?

So one can check it in python as such:

`from scipy.misc import imread`
`import numpy as np`
`img = imread("path/to/img.png")  # It loads the image as a numpy array.`
`print(np.count_nonzero(img)) # this gave me 0 indicating that all of the values are zeros` 

So the issue is either to do with AMD GPUs or Vulkan on Linux.  I do not know which and it will take me too long to figure out.

With an Nvidia 1080 TI it works fine, so I'm closing the issue.

closed 2019-02-27 by xksteven