carla-simulator/carla #1054: [0.9.1] Ubuntu16.04+Unreral4.19  make PythonAPI error

sheepwxy, 2018-12-17:
hello,when I execute the command "make PythonAPI" ,I got the error as belows:
*********************************************************************************************************************************
Setup.sh: llvm-5.0 already installed.
Setup.sh: boost-1.67.0 already installed.
Setup.sh: rpclib-d1146b7 already installed.
Setup.sh: googletest-1.8.0 already installed.
Setup.sh: Generating CMake configuration files.
Setup.sh: Success!
BuildLibCarla.sh: Building LibCarla "Client" configuration.
ninja: no work to do.
[1/1] Install the project...
-- Install configuration: "Client"
BuildLibCarla.sh: Success!
BuildPythonAPI.sh: Building Python API for Python 2.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing source/carla.egg-info/PKG-INFO
writing top-level names to source/carla.egg-info/top_level.txt
writing dependency_links to source/carla.egg-info/dependency_links.txt
reading manifest file 'source/carla.egg-info/SOURCES.txt'
writing manifest file 'source/carla.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
creating build/bdist.linux-x86_64/egg
creating build/bdist.linux-x86_64/egg/carla
copying build/lib.linux-x86_64-2.7/carla/__init__.py -> build/bdist.linux-x86_64/egg/carla
copying build/lib.linux-x86_64-2.7/carla/libcarla.so -> build/bdist.linux-x86_64/egg/carla
byte-compiling build/bdist.linux-x86_64/egg/carla/__init__.py to __init__.pyc
creating stub loader for carla/libcarla.so
byte-compiling build/bdist.linux-x86_64/egg/carla/libcarla.py to libcarla.pyc
creating build/bdist.linux-x86_64/egg/EGG-INFO
copying source/carla.egg-info/PKG-INFO -> build/bdist.linux-x86_64/egg/EGG-INFO
copying source/carla.egg-info/SOURCES.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying source/carla.egg-info/dependency_links.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
copying source/carla.egg-info/top_level.txt -> build/bdist.linux-x86_64/egg/EGG-INFO
writing build/bdist.linux-x86_64/egg/EGG-INFO/native_libs.txt
zip_safe flag not set; analyzing archive contents...
creating 'dist/carla-0.9.1-py2.7-linux-x86_64.egg' and adding 'build/bdist.linux-x86_64/egg' to it
removing 'build/bdist.linux-x86_64/egg' (and everything under it)
BuildPythonAPI.sh: Building Python API for Python 3.
setup.py:28: DeprecationWarning: dist() and linux_distribution() functions are deprecated in Python 3.5
  if platform.dist()[0].lower() in ["ubuntu", "debian"]:
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing source/carla.egg-info/PKG-INFO
writing dependency_links to source/carla.egg-info/dependency_links.txt
writing top-level names to source/carla.egg-info/top_level.txt
reading manifest file 'source/carla.egg-info/SOURCES.txt'
writing manifest file 'source/carla.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
building 'carla.libcarla' extension
clang-5.0 -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -Idependencies/include -I/usr/lib/gcc/x86_64-linux-gnu/7/include -I/home/wuxuyang/miniconda3/include/python3.7m -c source/libcarla/libcarla.cpp -o build/temp.linux-x86_64-3.7/source/libcarla/libcarla.o -fPIC -std=c++14 -Wno-missing-braces -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_ENABLE_LIFETIME_PROFILER -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
gcc -pthread -shared -B /home/wuxuyang/miniconda3/compiler_compat -L/home/wuxuyang/miniconda3/lib -Wl,-rpath=/home/wuxuyang/miniconda3/lib -Wl,--no-as-needed -Wl,--sysroot=/ build/temp.linux-x86_64-3.7/source/libcarla/libcarla.o -Ldependencies/lib -L/usr/lib/gcc/x86_64-linux-gnu/7 -o build/lib.linux-x86_64-3.7/carla/libcarla.cpython-37m-x86_64-linux-gnu.so /home/wuxuyang/simtools/carla/PythonAPI/dependencies/lib/libcarla_client.a /home/wuxuyang/simtools/carla/PythonAPI/dependencies/lib/librpc.a /home/wuxuyang/simtools/carla/PythonAPI/dependencies/lib/libboost_filesystem.a /home/wuxuyang/simtools/carla/PythonAPI/dependencies/lib/libboost_python37.a -lpng -ljpeg -ltiff /usr/lib/gcc/x86_64-linux-gnu/7/libstdc++.a
gcc: error: /home/wuxuyang/simtools/carla/PythonAPI/dependencies/lib/libboost_python37.a: No such file or directory
error: command 'gcc' failed with exit status 1
Util/BuildTools/Linux.mk:65: recipe for target 'PythonAPI' failed
make: *** [PythonAPI] Error 1
*********************************************************************************************************************
I try the commond "sudo apt-get install python-dev" ""sudo apt-get install python3-dev" ""sudo apt-get install libevent-dev"  ,but it does not help.

sheepwxy, 2018-12-17:
I found the error occurs when build with Python3

fabianoboril, 2018-12-17:
Hi,
it seems that you have different versions of Python3 on your system. When building libboost, Python 3.5 was used (coming from /usr/bin/python3), whereas, when building the PythonAPI, the build system tries to use Python 3.7 (coming from /usr/bin/env python3). 

You can test that by running
`/usr/bin/env python3`
and just
`/usr/bin/python3`
Most likely, the first one will report 3.7 as version, the second one 3.5.

If my guess is correct, you can try to remove the location of your local Python3.7 installation from your PATH environment. Does this work?

Best regards,

s7ev3n, 2018-12-21:
Hi, @fabianoboril 
I also have some issues with `make PythonAPI`, I have no clue, thanks

```
BuildLibCarla.sh: Success!
BuildPythonAPI.sh: Building Python API for Python 2.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing source/carla.egg-info/PKG-INFO
writing top-level names to source/carla.egg-info/top_level.txt
writing dependency_links to source/carla.egg-info/dependency_links.txt
reading manifest file 'source/carla.egg-info/SOURCES.txt'
writing manifest file 'source/carla.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
building 'carla.libcarla' extension
clang-5.0 -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -Wdate-time -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -Idependencies/include -I/usr/lib/gcc/x86_64-linux-gnu/7/include -I/usr/include/python2.7 -c source/libcarla/libcarla.cpp -o build/temp.linux-x86_64-2.7/source/libcarla/libcarla.o -fPIC -std=c++14 -Wno-missing-braces -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_ENABLE_LIFETIME_PROFILER -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
In file included from source/libcarla/libcarla.cpp:130:
In file included from source/libcarla/SensorData.cpp:9:
In file included from dependencies/include/carla/image/ImageIO.h:9:
In file included from dependencies/include/carla/image/ImageIOConfig.h:50:
In file included from dependencies/include/boost/gil/extension/io/png_io.hpp:35:
In file included from /usr/include/png.h:321:
/usr/include/pngconf.h:383:12: error: unknown type name '__pngconf'
           __pngconf.h__ in libpng already includes setjmp.h;
           ^
/usr/include/pngconf.h:383:21: error: cannot use dot operator on a type
           __pngconf.h__ in libpng already includes setjmp.h;
                    ^
/usr/include/pngconf.h:384:12: error: unknown type name '__dont__'
           __dont__ include it again.;
           ^
/usr/include/pngconf.h:384:28: error: expected ';' after top level declarator
           __dont__ include it again.;
                           ^
4 errors generated.
error: command 'clang-5.0' failed with exit status 1
Util/BuildTools/Linux.mk:65: recipe for target 'PythonAPI' failed
make: *** [PythonAPI] Error 1

```

s7ev3n, 2018-12-21:
After `make PythonAPI`, I also got a version mismatch issue which I describe in https://github.com/carla-simulator/ros-bridge/issues/13#issue-392947489.

sheepwxy, 2018-12-24:
> Hi,
> it seems that you have different versions of Python3 on your system. When building libboost, Python 3.5 was used (coming from /usr/bin/python3), whereas, when building the PythonAPI, the build [...]

yes

sheepwxy, 2018-12-24:
> Hi,
> it seems that you have different versions of Python3 on your system. When building libboost, Python 3.5 was used (coming from /usr/bin/python3), whereas, when building the PythonAPI, the build [...]

yes，It just works as you said ！！ thanks very much.

MichelangeloConserva, 2019-01-28:
```
gcc -pthread -shared -B /home/torquato/miniconda3/compiler_compat -L/home/torquato/miniconda3/lib -Wl,-rpath=/home/torquato/miniconda3/lib -Wl,--no-as-needed -Wl,--sysroot=/ build/temp.linux-x86_64-3.7/source/libcarla/libcarla.o -Ldependencies/lib -L/usr/lib/gcc/x86_64-linux-gnu/7 -o build/lib.linux-x86_64-3.7/carla/libcarla.cpython-37m-x86_64-linux-gnu.so /home/torquato/Desktop/carla_env/carla/PythonAPI/dependencies/lib/libcarla_client.a /home/torquato/Desktop/carla_env/carla/PythonAPI/dependencies/lib/librpc.a /home/torquato/Desktop/carla_env/carla/PythonAPI/dependencies/lib/libboost_filesystem.a /home/torquato/Desktop/carla_env/carla/PythonAPI/dependencies/lib/libboost_python37.a -lpng -ljpeg -ltiff /usr/lib/gcc/x86_64-linux-gnu/7/libstdc++.a
gcc: error: /home/torquato/Desktop/carla_env/carla/PythonAPI/dependencies/lib/libboost_python37.a: No such file or directory
error: command 'gcc' failed with exit status 1
Util/BuildTools/Linux.mk:65: recipe for target 'PythonAPI' failed
make: *** [PythonAPI] Error 1
```
```
which python
/home/torquato/miniconda3/bin/python
```
```
python
Python 3.7.2 (default, Dec 29 2018, 06:19:36) 
[GCC 7.3.0] :: Anaconda, Inc. on linux
```
Using latest version pulled from the master repository gives the same error.

MichelangeloConserva, 2019-01-29:
I managed to solve the issue through this steps.

##### Removing boost which gave the problem
1. rm -r Build/boost-*                         
##### Make python3.5 the default python out of conda environments
2. I deleted the default configuration of miniconda in ~/.bashrc using nano or gedit
##### Just making it again
3. make PythonAPI

code4meplz, 2019-05-09:
I had the same issue with 0.9.5 and UE 4.22 running Ubuntu 18.04
I ran the build with a conda python 3.7 env. Somehow it then fails to create the right `libboost_python37.a` files. I tried the above suggested methods and just reran `make PythonAPI` with a new conda 3.5 env. This doesnt work because the libboost files files are already missing. What worked for me was just deleting the whole carla folder, then creating a new conda environment with python 3.5 as default and building again. Should look something like this:
to create conda env:
`conda create --name carla_env python=3.5`
`conda activate carla_env`
then removing and rebuilding:
[u might want to add the UE path again for this new env : `export UE4_ROOT=~/UnrealEngine_4.22` or whereever ur UE lies and what version you are using]
`sudo rm -rf carla`
`git clone https://github.com/carla-simulator/carla`
`cd carla`
`./Update.sh`
`make launch`
`make PythonAPI`
`make package`

hope that helps someone :)

ShanshanYOU, 2019-05-10:
> Hi,
> it seems that you have different versions of Python3 on your system. When building libboost, Python 3.5 was used (coming from /usr/bin/python3), whereas, when building the PythonAPI, the build [...]

Hi, Exactly, from my part, i got 3.7 for the first one and 3.6 for the second one. But I have a lot of algos woking on both. Should I really remove one of them ?

Wh1teSh4dow, 2019-06-27:
I don't know if it helps anyone in this case but I've had similar errors due to the usage of anaconda which I managed to fix (namely, I wanted to create a virtual env using anaconda and work with the pythonAPI for carla and python3, under Ubuntu 18.04 LTS and UnrealEngine_4.22).

In order to start from a clean slate (especially if you've attempted other unsuccessful manipulations), when building carla on linux, you'll need to go to the root carla directory and execute
> make clean

then delete the build folder (which is left intact after make clean is executed
> rm -r Build/

Then, you have to remember the PythonAPI only support python27 and python35. What I did, which seemed to work is create a conda environment with python35 (which is the one I'm interested in using):
> conda create -n carla python=3.5
> conda activate carla

If I try to follow the instructions as is, however, this will not be enough. I run into issues relating to the inability to find pyconfig.h (but only for python35 for some reason). So I ended up following [this explanation here](https://stackoverflow.com/questions/54152667/how-to-fix-pyconfig-h-not-found-with-anaconda)  which overall just suggests to locate the pyconfig.h file for anaconda and run:
> export CPLUS_INCLUDE_PATH="$CPLUS_INCLUDE_PATH:path/to/your/Anaconda3/include/python3.7m"

(yes, for some reason, the 3.7 works).

I could then run (-j16 is for multithreading):
> make launch -j16

and then (you may attempt -j16 if you want)
> make PythonAPI
> make package -j16

And then I don't have any issues with compilation anymore.

sumedhpendurkar, 2019-09-17:
I tried cleaning the build and tried
> export CPLUS_INCLUDE_PATH="$CPLUS_INCLUDE_PATH:path/to/your/Anaconda3/include/python3.7m"

and then building again.
 I got carla/PythonAPI/carla/dependencies/lib/libboost_python35.a: No such file or directory

aurthes6220, 2019-09-20:
> I don't know if it helps anyone in this case but I've had similar errors due to the usage of anaconda which I managed to fix (namely, I wanted to create a virtual env using anaconda and work with [...]

I solve the problem with this way. Thank you very much.

ayadav10491, 2019-12-21:
Deactivating Conda base environment worked for me.

papabst, 2020-04-24:
I get these problems as well but already when making launch.
I'm running Ubuntu 18.04 LTS, 
no anaconda, 
/usr/bin/env python3 gives me 3.5, 
/usr/bin/python3 gives me 3.6
running `make launch` crashes on "Setup.sh: Extracting boost for Python 3" with "fatal error: 'pyconfig.h' file not found"
running `make PythonAPI` results in "gcc: error: /home/tum-vt/carla/PythonAPI/carla/dependencies/lib/libboost_python35.a: No such file or directory"
I have tried all kinds of paths for "export CPLUS_INCLUDE_PATH="$CPLUS_INCLUDE_PATH" but nothing seems to work

I have no idea what to do anymore. I'd favor to build Carla with python3.5 and the sole purpose of the computer I'm working on is to run Carla. Can you please help me

KShah707, 2020-04-29:
Also having the same issue, having tried all the same steps above^

papabst, 2020-04-29:
I was now able to fix this by taking a closer look at the errors that are thrown when pyconfig.h could not be found. In my case clang was looking at the wrong path. It looked for pyconfig.h in usr/include/python3.5 while I only had a usr/include/python3.5**m** directory. My very ugly solution was to rename the folder and create a link to it at the same place with the old name. 

My guess is that there should be a clang config file where one could change the path for python3 but I have not looked into it.

KShah707, 2020-04-30:
@papabst Did you get past the issue with `make PythonAPI`? I was also able to get `make launch1` working by adding the anaconda `python3.5m` to the `CPLUS_INCLUDE_PATH` but it still didn't help with the `libboost_python35.a` error. Weirdly enough the setup script runs through without (noticeable) errors and compiles boost for python 2.7 but not 3.5.

therobotprogrammer, 2020-07-26:
@papabst 

Here is my solution. It seems the problem occours when using anaconda: 

#Problem:
gcc: error: _[carla_repository_folder]_/carla/PythonAPI/carla/dependencies/lib/libboost_python37.a: No such file or directory
error: command 'gcc' failed with exit status 1
Util/BuildTools/Linux.mk:83: recipe for target 'PythonAPI' failed
make: *** [PythonAPI] Error 1

#Solution:
cd ~/anaconda3/envs/_[your_anaconda_enviroenment_name]_/lib
cp libboost* _[carla_repository_folder]_/carla/PythonAPI/carla/dependencies/lib

TrinhTuanHung2021, 2021-08-12:
> After `make PythonAPI`, I also got a version mismatch issue which I describe in [carla-simulator/ros-bridge#13 (comment)](https://github.com/carla-simulator/ros-bridge/issues/13#issue-392947489).

So I am using Win 10. So could you tell me how to "make PythonAPI" in the command line? It shows me" make is not recognized as an internal or external command".
Thanks