carla-simulator/carla #1055: Getting lidar point cloud data from Carla

pfrcks, 2018-12-17:
I am trying to capture the `sensor.lidar.ray_cast` as mentioned on the [page](https://github.com/carla-simulator/carla/blob/master/Docs/cameras_and_sensors.md), however on visualizing the pointcloud it appears to be too sparse. Any suggestions as to how to improve the quality of the data captured?

analog-cbarber, 2018-12-17:
Not sure what you mean by "too sparse". Do you mean you cannot see the points very well? If so, first try rendering your point cloud using larger points.

pfrcks, 2018-12-17:
Hey,

Thanks for the response. I am able to see the points but they are not dense, i.e, they are not able to capture the scene with decent quality.

[image]

closed 2018-12-19 by pfrcks