carla-simulator/carla #1064: Can i extract point cloud from depth map image in the version 0.9.x?

1453042287, 2018-12-20:
Hello developers and community,
like the title, i used the point_cloud_example.py in the stable version to generate the point_cloud file, but can i use the same method in version 0.9.x?

zrfreya, 2018-12-20:
I have the same problem. I tried to use the point_cloud_example to project the pixels in the semantic labels to 3d space, then compared the new locations with cars' bounding boxes to get the instance labels.  But I can't get the right 3d positions.

zrfreya, 2018-12-22:
> I have the same problem. I tried to use the point_cloud_example to project the pixels in the semantic labels to 3d space, then compared the new locations with cars' bounding boxes to get the instance labels. But I can't get the right 3d positions.

I found that if I change the to_unreal_transform to yaw = -90,roll=90, then the result is correct

1453042287, 2018-12-23:
@zrfreya good news:)

closed 2019-02-28 by stale[bot]