carla-simulator/carla #1065: Question about autopilot control in 0.9.1

chinen95, 2018-12-20:
Is there any way to get autopilot control from server without apply them in 0.9.1. Just like pythonclient did in 0.8.x.

fabianoboril, 2018-12-21:
Hi,
can you please elaborate a little bit, what you want to do? To activate/deactivate the autopilot you can use the PythonAPI via set_autopilot():
https://carla.readthedocs.io/en/latest/python_api/

Regards

chinen95, 2018-12-23:
> Hi,
> can you please elaborate a little bit, what you want to do? To activate/deactivate the autopilot you can use the PythonAPI via set_autopilot():
> https://carla.readthedocs.io/en/latest/python_api/
> 
> Regards

Hi, thanks for your reply.
I wanna to add some noise to built-in autopilot control for dataset aggregation and overwrite steer control to change lane manually. built-in autopilot control + (noise or manual control) → output control

closed 2019-02-28 by stale[bot]

eslambakr, 2019-06-11:
Hello,
I have the same problem with Carla version 9.x (specially 9.5).

I also tried to apply the noisy controls while using the auto pilot but It didn't work.

So I think we need to get the controls of the auto pilot without applying it.

Do you have a solution for it?

Thanks in advance.