carla-simulator/carla #1068: std::bad_alloc

s7ev3n, 2018-12-21:
Hi,
I am using carla 0.9.1, and I run `./CarlaUE4.sh  -carla-server -windowed -ResX=320 -ResY=240` which launches Town3, and i dont see 'Waiting for the client to connect... '.
```
[2018.12.21-07.23.53:948][  0]LogLoad: (Engine Initialization) Total time: 10.50 seconds
[2018.12.21-07.23.54:014][  0]LogRenderer: Reallocating scene render targets to support 320x240 Format 10 NumSamples 1 (Frame:1).
[2018.12.21-07.23.54:584][  1]LogLinux: Setting swap interval to 'Immediate'
[2018.12.21-07.24.08:845][674]LogCarla: Starting AWorldObserver sensor

```
When I `python spawn_npc.py -n 80`, it got an error:
```
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
```
ANY idea??

s7ev3n, 2018-12-21:
When start carla, also have some errors:
```
[2018.12.21-07.48.54:291][  0]LogLinux: Selected Device Profile: [LinuxNoEditor]
[2018.12.21-07.48.54:352][  0]LogNetVersion: CarlaUE4 , NetCL: 0, EngineNetVer: 2, GameNetVer: 0 (Checksum: 439411406)
[2018.12.21-07.48.54:534][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapRoad requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapRoad
[2018.12.21-07.48.54:534][  0]LogStreaming: Error: Found 0 dependent packages...
[2018.12.21-07.48.54:534][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapSideWalkCurb requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapSideWalkCurb
[2018.12.21-07.48.54:534][  0]LogStreaming: Error: Found 0 dependent packages...
[2018.12.21-07.48.54:534][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapSideWalk_00 requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapSideWalk_00
[2018.12.21-07.48.54:534][  0]LogStreaming: Error: Found 0 dependent packages...
[2018.12.21-07.48.54:535][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapLaneMarking requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/Ground/SimpleRoad/CheapLaneMarking
[2018.12.21-07.48.54:535][  0]LogStreaming: Error: Found 0 dependent packages...
[2018.12.21-07.48.54:535][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/WetPavement/WetPavement_Complex_Road_N2 requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/WetPavement/WetPavement_Complex_Road_N2
[2018.12.21-07.48.54:535][  0]LogStreaming: Error: Found 0 dependent packages...
[2018.12.21-07.48.54:535][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/WetPavement/WetPavement_Complex_Concrete requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/WetPavement/WetPavement_Complex_Concrete
[2018.12.21-07.48.54:535][  0]LogStreaming: Error: Found 0 dependent packages...
[2018.12.21-07.48.54:536][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/Ground/SideWalks/SidewalkN4/WetPavement_SidewalkN4 requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/Ground/SideWalks/SidewalkN4/WetPavement_SidewalkN4
[2018.12.21-07.48.54:536][  0]LogStreaming: Error: Found 0 dependent packages...
[2018.12.21-07.48.54:536][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Static/GenericMaterials/LaneMarking/Lanemarking requested by async loading code. NameToLoad: /Game/Carla/Static/GenericMaterials/LaneMarking/Lanemarking
[2018.12.21-07.48.54:536][  0]LogStreaming: Error: Found 0 dependent packages...
[2018.12.21-07.48.54:536][  0]LogCarla: == CARLA Settings ==============================================================

```

fabianoboril, 2018-12-21:
Hi,
you get this error when you run the Client using a PythonAPI version that is not compatible with the server. It seems that you use the 0.9.1 pre-compile release for the server, but for the client it seems you compile your own Carla (I assume from master branch). You should use this version also to start the server.

s7ev3n, 2018-12-21:
Hi, @fabianoboril
I do use the 0.9.1 pre-compiled release server. But I would like to use the official ros_bridge with pre-compiled carla. 
I followed the readme in ros_bridge, which has to install carla python client API in ros workspace which requires to compile carla. 
```
cd carla/PythonAPI
pip2 install -e .  --user --upgrade
```

Does it implies that I have to compile my own carla binary to work with ros_bridge??

fabianoboril, 2018-12-21:
Hi,
well this problem is different from your question that your originally posted. 

The ros-bridge was never working with 0.9.1, as 0.9.1 was release before we started to adapt the ros-bridge to the new python API of CARLA. However, you can download a pre-compiled version that should work with the ros-bridge here:
http://158.109.9.218:8080/blue/organizations/jenkins/carla/detail/master/12/artifacts
I haven't tested it with this version, but it is worth a try.

Once you have this version installed, then take a look here:
https://github.com/carla-simulator/ros-bridge/issues/14#issuecomment-449347438
Pip install is used, when you build from source, not when you use the compile egg-files. In this case, you can use easy_install or simply extend your PYTHONPATH env variable.

Regards

closed 2018-12-24 by s7ev3n