carla-simulator/carla #1069: worse graphic rendering quality in 0.9.1 compared to 0.8.4

AlenUbuntu, 2018-12-21:
Hello:
Carla is wonderful simuator for autonomous driving. However, I found that the graphic rendering quality of Carla 0.9.1 is significantly worse than that in Carla 0.8.4. I have tried to run both simulators on the same machine (GTX 1060) and found that the rendering quality is significantly worse in Carla 0.9.1. 
Here is the example:
[image]
[image]

I have checked the GPU usage. Carla 0.8.4 is capable of using 100% GPU while Carla 0.9.1 uses only 33% GPU with most resources unoccupied.

Is there any way to improve the rendering quality in Carla 0.9.1? (I have tried Epic mode and it does not work)

analog-cbarber, 2018-12-21:
Would be better if you showed the same scene or at least the same town.

AlenUbuntu, 2018-12-21:
How to change the map? The command in the tutorial seems not working (there is no such files in that directory). I use the binary version of carla 0.9.1. Could you help me with it?

moenasrallah, 2018-12-21:
@AlenUbuntu 
To be honest, the bottom picture (0.91) is better quality than the top one (0.84).

Rendering is defined as so: "the processing of an outline image using color and shading to make it appear solid and three-dimensional."

"Rendering" in 0.84 is blurrier. Although you may think its closer to realism, it is not a factor of better performance. In 0.91, it is more optimized and clearer. 0.84 ran slower and the graphics were not optimized from a game development and simulation point of view.

You can even see it in the text on-screen. Of course... I'd leave it to the Carla team to give a more detailed explanation.

I would go through their documentation about changing the map btw... their website gives good info about the simulator
https://carla.readthedocs.io/en/latest/configuring_the_simulation/#changing-the-map

nsubiron, 2018-12-22:
Hi @AlenUbuntu,

Thanks for the observation! We are aware the visuals have changed in latest versions, and we'll work to improve the visual aspect of 0.9.x versions.

There are however a few differences in those pictures that are not only due to this. The upper picture is taken from the simulator window while the bottom one is taken from the client-side, those are different types of cameras in UE4 with slightly different render pipelines. The capture cameras miss a few final post-process effects that make the image looks nicer, unfortunately these are the only cameras in UE4 we've found that allow us capturing images and sending them to the client. The other difference is that the second picture is taken in the new (bigger) city, this city has been more optimized for performance at the cost of loosing some visual "realism". There is always a trade-off between better graphics and performance. 

But as you said, the visual feeling (call it quality, realism, ...) has changed. We plan to work on this in the coming year!

Jack-Pinnington, 2019-12-15:
Hello,

Is there any update regarding this issue? I'm using Carla for Computer Vision testing and would like to find a way to improve my graphics quality - as you said, the realism. 

I'm creating sensors from the blueprint library and attaching them to my car:

_camera_bp = blueprint.find('sensor.camera.rgb') #default set to rgb
self.sensor = world.spawn_actor(camera_bp, camera_transform, attach_to=car)__

Is there any way that my cameras could be set to "PhotoReal" as well? Not bothered about running in real-time since it's all for Computer Vision anyway. Would ideally like as many Post-Processing effects as possible! Is this done through Carla directly or through Unreal?

Thanks,
Jack

germanros1987, 2020-03-19:
This issue is now definitely solved. As proof:

[image]

closed 2020-03-19 by germanros1987

Jack-Pinnington, 2020-03-20:
Hello,
I improved my graphics by following Issue 2019:
https://github.com/carla-simulator/carla/issues/2019

Hopefully that helps others like it helped me.

Thanks,
Jack