carla-simulator/carla #1072: [0.9.1][Ubuntu 16.04] How to get the actors' transform synchronized with the sensor image

zrfreya, 2018-12-23:
I am trying to get the instance segmentation label by using the semantic segmentation label and the depth map.  I do the following steps:
1. Add depth and semantic segmentation sensors in manual_control.py
2. Add a save_position function to save the camera transform, the transforms and actor ids of all cars and  the bounding boxes of all car in _parse_image() function.
3. Convert the semantic segmentation label to 3d point cloud in world coordinate system. For points of cars, compare their 3d location with 3d bounding boxes of all cars. If one point is in the bounding box of the Nth car, set the corresponding pixel to N.

However in my results, there may be some instances truncated.
[image]

[image]

I found that in the save_position function of step 2 when I call the get_transform function of the same car twice, the transfroms are different. That means when the get_transform function is called, the transforms of the cars have been changed. 

I wonder if there is a way to get the actors' transform synchronized with the sensor image ? Look forward to replies. Thank you.

1453042287, 2018-12-24:
i believe the best way is to use the G&B channel to record the instanse ID when generating the semantic segmentation labels, but for now, i have no ideas how to modify the code :(

1453042287, 2018-12-24:
and the walker's 3d bounding box is not accurate :(

fabianoboril, 2018-12-24:
Hm... I am not sure, if there is an out-of-the-box function for that, but that may be a good addition @nsubiron?

What I would suggest for now:
You can use the PythonAPI world.on_tick() function, to get the transform of your vehicle on every world tick. In addition, you get the timestamp and frame-id for this transform with this callback. In addition, you get your images, which also contain a frame-id. Now, you can match the ids and you should have the transform that corresponds to the image.

Merry XMas!

zrfreya, 2018-12-24:
@fabianoboril  Thank you for your advice. I will try it. Merry Chrismas ~

nsubiron, 2019-01-09:
The transforms and the images come asynchronously on different messages from the server. The transforms are sent during the UE4 Actor's update step, and the images after the render step, this can be a couple of updates later.

At this point the best we have is what @fabianoboril suggests. Note however that nothing guarantees that you would not get a new update half-way parsing the transforms, though it's unlikely given the speed of the server. I agree there should be a more robust / easy to use method in our API.

closed 2019-03-17 by stale[bot]

benkmoore, 2019-07-09:
@zrfreya would you mind sharing your code so I can see how you achieved this? thanks