carla-simulator/carla #1080: Using the Scenario_Runner in Carla 0.9.2

ndeshp2s, 2018-12-27:
Thanks for the new release of CARLA with such wonderful features.

I have been trying to test the Scenario_Runner, but not sure with to expect as the output. In the terminal (where I run the command say "python scenario_runner.py --scenario FollowLeadingVehicle") it says the concerned scenario is running. However, I do not see any new window neither do I see anything happening at the server end. I have followed the exact steps mentioned in the documentation. 

Thank you.

mohit-awana, 2018-12-27:
@ndeshp2s you need to run manual_control .py script concurrently to see and run the scenario.

ndeshp2s, 2018-12-27:
@mohit-awana Thanks for your quick reply. I did try running the manual_control.py script. And it seems to be working now. Although it did not work for "OppositeVehicleRunningRedLight". But I will recheck for this particular scenario.  

Thanks a lot once again.

closed 2018-12-27 by ndeshp2s

antoniodvc, 2019-01-11:
Hi @ndeshp2s, how do you change default map "Town03" in Carla 0.9.2? I see "FollowLeadingVehicle" scenario only works with Town01.

Thank you.