carla-simulator/carla #1082: OpenDrive precision

Unlingius, 2018-12-27:
I try to use Opendrive file for getting ground truth 3d track of lane markings. What I did:
1) select position of camera
x, y, z = -86.65433502, -160.461639404, 1.01022005081
roll = -0.0714416503906, pitch = -0.129336461425, yaw = 92.9592056274
2) set camera simulate physics to false and set camera location to that point and angles
3) parse opendrive file to get geometry of road markings
4) draw current segment markings on segmentation image
result:
[image]
As you see, it is not ideal. Or is it normal? What precision between Town3 mesh and corresponding OpenDrive file? What precision is function "set_camera_transform"?

YashBansod, 2019-01-04:
Hello @Unlingius I wanted to know how you did step 3 and 4 that you have mentioned i.e. 
- how did you parse the opendrive file to get geometry of road markings?
- how did you draw the current segment markings from the opendrive file onto the segmentation image?
    - did it require you to transfrom the lane coordinates to camera frame coordinates etc.

marcgpuig, 2019-01-08:
Hi @Unlingius.
Thanks for your report.

To clarify, the OpenDrive `.xodr` file is directly generated from RoadRunner and it should be ok. We do not had problems with `set_camera_transform` previously. Also, the roads are well positioned on the world map, @XGodina just checked it. Are you sure the results are not good?

I will take a look at it in a few days.

Unlingius, 2019-01-09:
@marcgpuig    elevation profile of my road is:
<elevationProfile>
            <elevation s="0.0" a="0.0" b="0.0" c="0.0" d="0.0"/>
</elevationProfile>
Is it correct, that z-coordinate of marking in Carla is zero?

Unlingius, 2019-01-09:
to be sure, that i am not wrong in rotations order, i try all the same operations using point x,y,z,roll,pitch,yaw = -81.0, -141.0, 0.0, 0.0, 90.0
and got result:
[image]
in the right down part 6 pixels difference

XGodina, 2019-01-09:
Hi @Unlingius 

The z-coordinate of markings is a bit up respect to road for avoid z-fighting.

Unlingius, 2019-01-09:
@XGodina  Oh, possible this is my problem. Thank you. How can I know this 'bit up"? is it the same for all roads?

Unlingius, 2019-01-10:
I don't know, what is the problem, but when i tried all the same on custom map from new version of roadrunner, it looks nice.

XGodina, 2019-01-15:
@Unlingius The Town03 was made with an old version of RoadRunner and does not include the 
option of "LaneMarkings separated from the highway". It had to be done manually. In the current versions is fixed.

aravindSwamy94, 2019-02-18:
Hi , Just few question regarding this. 
    1. I suppose this opendrive map will have data for the entire map.If so , how do you recognise the points that are ahead of your vehicle and map it to your segmented images?
    2. Also parsing of opendrive file means, do you parse all the lane geometry points of the map at one stretch or just do it only for a certain distance.

Please help me on this regard.

Unlingius, 2019-02-18:
@aravindSwamy94 1) yes. But we have car position(or camera position) from carla in python examples. FIrst, i check ['road']['planView']['x'] and ['y'] to find closest segment to my position. Then i check ['road']['link'] to find connected segments.
2) At start, i load whole file with python xml library. Opendrive file looks in my program as a cascade of dicts and lists. But rasterization (generaration a set of 3d points) for segment i do only for closest segment and connected. After that, it is possible to cut the result by max distance if you want.

silgon, 2019-03-29:
@Unlingius 
What are you using to visualize the xord file? Did you look into the documentation parse everything yourself or did you find a library?

Unlingius, 2019-04-01:
@silgon  Hi! I parsed it myself.

closed 2019-06-07 by stale[bot]

zkyxsq, 2020-06-12:
> I don't know, what is the problem, but when i tried all the same on custom map from new version of roadrunner, it looks nice.

hello, how to get roadrunner license?

zkyxsq, 2020-06-12:
> > I don't know, what is the problem, but when i tried all the same on custom map from new version of roadrunner, it looks nice.
> 
> hello, how to get roadrunner license?

free roadrunner license,thank you.

yuzehh, 2022-11-25:
> I try to use Opendrive file for getting ground truth 3d track of lane markings. What I did:
> 
> 1. select position of camera
>    x, y, z = -86.6543, 160.4616, 1.0102
>    roll = -0.07144, pitch = -0.1293, yaw = 92.95920
> 2. set camera simulate physics to false and set camera transform to that point and angles
> 3. parse opendrive file to get geometry of road markings
> 4. draw current segment markings on segmentation image
>    result:
>    [image]
>    As you see, it is not ideal. Or is it normal? What precision between Town3 mesh and corresponding OpenDrive file? What precision is function "set_camera_transform"?

Hi, I recently have exactly the same needs as you are, i.e., get the GT of lane markings. Do you have any plans to public your code OR could you please share your code with me? I think it will be popular. Thanks~

zkyxsq, 2022-11-25:
Thanks for your letter. I will deal with it quickly. Yours,Kunyu ZhangHarbin Engineering University,China