carla-simulator/carla #1083: ImportError: No module named derived_object_msgs.msg

JackRen88, 2018-12-28:
Hello，everyone
I download the prebuilded  version of Carla ,and I completed to  Install the CARLA Python API according to the guide(https://github.com/carla-simulator/ros-bridge),runnning this:
python -c 'import carla;print("Success")' ,Show success message without any error.
But when I run:
roslaunch carla_ros_bridge client.launch 
there is the following error:
`ImportError: No module named derived_object_msgs.msg`
How can I solve this problem?

fabianoboril, 2018-12-28:
Hi,
you need to install the derived-object-sensor package for ROS, e.g. something like
ˋˋˋ
apt-get install ros-kinetic-derived-object-sensor-msgs
ˋˋˋ

JackRen88, 2018-12-28:
@fabianoboril 
Thanks！

closed 2019-01-14 by marcgpuig

nishanthjois, 2019-01-25:
@JackRen88 Did below command work? Its's not for me.
apt-get install ros-kinetic-derived-object-sensor-msgs

raphaellmsousa, 2019-02-02:
Hi guys, I'm having the same problem here. Did you guys find a solution? Thanks..

berndgassmann, 2019-02-04:
The solution is listed above and therefore the issue is closed.
Did you install those:

apt-get install ros-kinetic-derived-object-sensor-msgs

zhenwei01, 2021-07-16:
if you Unable to locate package ros-kinetic-derived-object-sensor-msgs, try this 

apt-get install ros-kinetic-derived-object-msgs

wuxiaohua1011, 2022-04-12:
in foxy, it is `sudo apt install ros-foxy-derived-object-msgs`