carla-simulator/carla #1090: How to get Roadrunner?

JackRen88, 2019-01-02:
Hello,carla team.
I are trying to use Roadrunner of Vector Zero to create my own map, how  should I download a trial of RoadRunner?

JackRen88, 2019-01-02:
@wlemkens ,Hi.
Do you requset a trail? I request a trial,but now VectorZero haven't responded to me yet.

wlemkens, 2019-01-02:
It took some time before they respond to me as well.

Soolek, 2019-01-02:
I have the same problem - only a 14 day evaluation license. I did ask about the license "for academic and research purposes to those academics who request it", but so far now I got no response.

wlemkens, 2019-01-07:
I got the message that the he license "for academic and research purposes to those academics who request it" is the time limited one.
It reduces the usefulness considerably in my book. 

To me, it looks more like they are using Carla as a marketing tool than as a support for research. But I'll happily be proven wrong by them.

XGodina, 2019-01-07:
Hi @JackRen88 @wlemkens @Soolek, We have contacted directly with VectorZero. We'll be reporting on how the issue evolves. 

Sorry for the inconvenience

zlacelle, 2019-02-04:
I'd also like to see a solution to this. I've found only 3 OpenDRIVE editors so far: VectorZero's RoadRunner ($$$), VIRES ROD ($$$), and OpenRoadEd (unsupported and very light in features).

XGodina, 2019-02-05:
Hi, @heethesh @Soolek @wlemkens @JackRen88 we talked with VectorZero. The last update we know so far. They working on a better method to support academically.  Now they have started to create a floating license server for universities. 

We'll be reporting

gaetanlefrioux, 2019-03-21:
Hi, any fresh news from VectorZero?

zlacelle, 2019-06-20:
Adding activity because we still need access to RoadRunner, or another way to generate OpenDRIVE maps.

kbarbora, 2019-06-22:
Yes, we need the Roadrunner software. Adding activity.

closed 2019-08-28 by stale[bot]

basavaraj1995, 2019-12-16:
Hi, any fresh news from VectorZero? @XGodina

EsquilliDev, 2020-01-16:
To follow up with this, any alternative to RoadRunner?

YDSh510, 2020-02-05:
Hi Carla Team,

Any update from Roadrunner on licensing part. Else do we have any other alternate way to generate OpenDRIVE maps. I think many people are struck because of that..

hbanzhaf, 2020-03-27:
There is an open-source OpenDrive editor called [OddLOT](https://www.hlrs.de/solutions-services/service-portfolio/visualization/driving-simulator/oddlot/). It is contained within the [COVISE repository](https://github.com/hlrs-vis/covise). I compiled COVISE along with OddLOT and imported one of the [OpenDrive examples](http://opendrive.org/download.html):

[image]

If you also would like to compile COVISE/OddLOT, make sure that it finds [OpenCRG](http://opencrg.org/) (along with the other required packages), otherwise, OddLOT will not be built (see this [issue](https://github.com/hlrs-vis/covise/issues/44)). Let me know if you need any assistance setting up OddLOT. If there is enough interest, we could also share a Docker container with OddLOT already compiled.

pierluigiolleja, 2020-05-08:
> There is an open-source OpenDrive editor called [OddLOT](https://www.hlrs.de/solutions-services/service-portfolio/visualization/driving-simulator/oddlot/). It is contained within the [COVISE [...]

Hi, thank you for your suggestion. 
Is there a possibility to get COVISE/OddLOT already compiled? 

Thank you

hbanzhaf, 2020-05-08:
I found this [page](https://fs.hlrs.de/projects/covise/support/download/) where you can download already compiled versions of COVISE. However, I have never tried them and I do not know whether they include OddLOT.

MauFranco, 2020-06-18:
Hello All,
I'm also interested in getting a compiled version of OddLOT.  Please share it if you have been successful compiling it.
Thanks a lot in advance,

DevotionGit, 2020-08-21:
> There is an open-source OpenDrive editor called [OddLOT](https://www.hlrs.de/solutions-services/service-portfolio/visualization/driving-simulator/oddlot/). It is contained within the [COVISE [...]

Hi,could you share a Docker container with OddLOT already compiled?
Thank you !

tranqkhue, 2020-09-06:
> There is an open-source OpenDrive editor called [OddLOT](https://www.hlrs.de/solutions-services/service-portfolio/visualization/driving-simulator/oddlot/). It is contained within the [COVISE [...]

Hi, I'm interested in OddLOT too. May you share the docker container?

hbanzhaf, 2020-09-06:
Thank you for your interest. Please find all further information [here](https://github.com/hbanzhaf/docker_covise).

uwewoessner, 2020-09-06:
> Thank you for your interest. Please find all further information [here](https://github.com/hbanzhaf/docker_covise).

Thank you for the docker container!
And to everyone who uses OddLOT: don't hesitate to contact me if you are having issues with it or if you have a feature request.
We are currently reworking the interaction concept and it might not work as expected, thus please help us to improve it.
PS.: the windows binaries from here: https://fs.hlrs.de/projects/covise/support/download/ do include OddLOT.

tranqkhue, 2020-09-06:
> Thank you for the docker container!
> And to everyone who uses OddLOT: don't hesitate to contact me if you are having issues with it or if you have a feature request.
> We are currently reworking the [...]

I have just installed the windows binary version. The installation was really straightforward and I successfully ran the program. However, when I tried to import OSM map, the program crashes. Is there a solution for this problem? Many thanks :D

uwewoessner, 2020-09-07:
> I have just installed the windows binary version. The installation was really straightforward and I successfully ran the program. However, when I tried to import OSM map, the program crashes. Is there a solution for this problem? Many thanks :D
Could you send me the OSM file so that I can debug it?
It might be the projection settings, the default is Gauss Krüger which might not be OK for you anyway. If the projection is wrong all coordinates end up in Nirvana and road creation might not work. Also check the X and Y offsets on the same setting page.
You can also open a COVISE Shell and start oddlot from the command line, then you might see some error message.

tranqkhue, 2020-09-07:
> Could you send me the OSM file so that I can debug it?

Many thanks  :D My OSM file was downloaded from openstreetmap.org
[map.zip](https://github.com/carla-simulator/carla/files/5181581/map.zip)
Also, I opened oddlot from command line, and it's really weird that there was no debug information available.
[image]

uwewoessner, 2020-09-07:
please try out this version:
https://fs.hlrs.de/projects/covise/support/download/snapshot-2020-09-07/covise_2020-09-07_zebuopt.exe
after File->New go to File->Settings and select the "UTM Zone 45" preset. Then you can load your OSM file.

tranqkhue, 2020-09-08:
> please try out this version:
> https://fs.hlrs.de/projects/covise/support/download/snapshot-2020-09-07/covise_2020-09-07_zebuopt.exe
> after File->New go to File->Settings and select the "UTM Zone 45" preset. Then you can load your OSM file.

Many thanks :D It works.

uwewoessner, 2020-09-08:
If there is enough interest in the carla community I could give a webex presentation on how to use OddLOT and perhaps find some developers who would like to help developing.

tranqkhue, 2020-09-08:
> If there is enough interest in the carla community I could give a webex presentation on how to use OddLOT and perhaps find some developers who would like to help developing.

I'm in :D Currently, up to my knowledge, there is no open-source software that could help to build the OpenDrive map for simulator except OddLOT. OpenDrive is an open standard, but not many open editors! Of course, I will cite OddLOT in my research!

kanakab, 2020-09-11:
@uwewoessner That would be great. I would like to contribute too. I think lot of people are tired of not having good OpenSource software to create maps.

kanakab, 2020-09-11:
@uwewoessner @hbanzhaf  I tried using installing using the docker file you mentioned. I am getting the following error, please help me rectify it.
[image]

mikelgarciafonseca, 2020-09-15:
> If there is enough interest in the carla community I could give a webex presentation on how to use OddLOT and perhaps find some developers who would like to help developing.

I would love to see that presentation

Basavaraj-PN, 2020-09-15:
> If there is enough interest in the carla community I could give a webex presentation on how to use OddLOT and perhaps find some developers who would like to help developing.

That would be great. 👍

licj15, 2020-12-17:
> If there is enough interest in the carla community I could give a webex presentation on how to use OddLOT and perhaps find some developers who would like to help developing.

licj15, 2020-12-17:
Looking forward to this presentation! Do you need any help to organize it?

> If there is enough interest in the carla community I could give a webex presentation on how to use OddLOT and perhaps find some developers who would like to help developing.

uwewoessner, 2020-12-17:
I am busy till Tuesday.
Not sure whether a Webex before Christmas or between Christmas and New Year would be such a good idea. How about early January?

licj15, 2020-12-18:
@uwewoessner 
Thanks!
(1) Before Christmas (2) between Christmas and New Year (3) Early January (Jan. 1 ~ 10). All the 3 options work for me perfectly. How about others who are interested in OddLOT?

If possible, people who are interested in it can fill this survey below to find a common time? 
[when2meet](https://www.when2meet.com/?10547446-uYcVy) (Everyone can see the result)

uwewoessner, 2020-12-18:
Thank you for the poll. Would you please add January 12 and 13 to the list? Everyone in Germany is on holiday up until January 10

licj15, 2020-12-18:
> Thank you for the poll. Would you please add January 12 and 13 to the list? Everyone in Germany is on holiday up until January 10

@uwewoessner 
Thanks for sharing this information. I have updated the link to [when2meet](https://www.when2meet.com/?10547568-KbtfB) with Jan 11 - 14 added. But the information you have filled in before may has been lost, could you fill it again. Thanks!

Also, @Basavaraj-PN @mikelgarciafonseca, I think you are also interested in this event based on the history of the issue. Would you like to join the Webex?

Basavaraj-PN, 2020-12-19:
@licj15 sure, thank for invitation.

hs4-og, 2021-01-12:
> Thank you for the poll. Would you please add January 12 and 13 to the list? Everyone in Germany is on holiday up until January 10

I would love to join the presentation as well.

uwewoessner, 2021-01-12:
Then Let's do a small webex tomorrow night at 7PM German time 
https://unistuttgart.webex.com/unistuttgart/j.php?MTID=m90355035c067f92aa7f863ea15161073

amzoo, 2021-01-28:
Did anyone happen to record the session? Would love to have attended.

amzoo, 2021-01-28:
To those inquiring about RoadRunner, MathWorks (MatLab parent company), absorbed VectorZero mid 2019. RoadRunner is now packaged inside of MatLab. From the sales department:

> RoadRunner products are not available on the Home License. In this case, a Standard License would be needed to purchase that product. You can request a quote for RoadRunner from the link: https://www.mathworks.com/pricing-licensing.html

I'm inquiring about pricing now, but for students you can get a license through your university. Have confirmed this works with one of my colleagues who is still an enrolled student.

augri, 2021-05-19:
@uwewoessner  Do you plan any other presentations on OddLot in the near future? Or do you have any useful material to help use it?

darld, 2021-06-21:
@uwewoessner Do you plan any documentation for OddLot? I would love to watch your presentation record if have.

darld, 2021-06-21:
Maybe this can be create CARLA map. https://github.com/johschmitz/blender-driving-scenario-creator

Laudarisd, 2021-07-06:
Vectorzero Roadrunner doesn't give any response. @uwewoessner  can I import shp and osm files in ODDLOT ?  I have to make my own map for carla. Totally stuck here.

uwewoessner, 2021-07-06:
Oddlot can import osm files but no shp files. If shape files are much better for you I can give it a try and implement an importer for shape files.

Laudarisd, 2021-07-06:
Thank you for the quick reply, @uwewoessner let me try ODDLOT , I have a question? Does it export osm in xodr(opendrive format)?

uwewoessner, 2021-07-06:
yes, it can import osm and a few other tile formats and export xodr.

Laudarisd, 2021-07-06:
@uwewoessner  Hi, sorry for bothering you, I was compiling ODDLOT , while running make got following error. 
Any idea 
```
Could not find a package configuration file provided by "boost_chrono"
  (requested version 1.71.0) with any of the following names:

    boost_chronoConfig.cmake
    boost_chrono-config.cmake
```
Thank you

uwewoessner, 2021-07-06:
you have to install the boost-devel package, or if you compile boost, you should compile with --with-chrono

raul-parada, 2022-04-26:
Does someone know any tutorial to make OpenCover work? I've installed successfully, however, I don't know how to start creating the environment. I'm using the below version on Windows.

https://fs.hlrs.de/projects/covise/support/download/snapshot-2022-03-20/#top

gouwylieven, 2023-05-04:
Hi me too I am desperate about getting a roadrunner license but getting no response and being an individual without a company connection and without being a student, they simply don't reply. Does anybody have any experience with the Truevision Designer mentioned by Carla? Does it support importing GIS data?

frederiknolte, 2023-05-04:
I tried Truevision Designer and it was not functioning. I just could not figure out how to do even the basics such as connecting two roads as their "documentation" (Highlighted information when hovering over buttons) did not give the correct commands either. This was on both Windows and Ubuntu. I now have a RoadRunner license via my university and gave up on Truevision Designer.

gouwylieven, 2023-05-04:
Aw that's sad. Unfortunately I am not 

> I tried Truevision Designer and it was not functioning. I just could not figure out how to do even the basics such as connecting two roads as their "documentation" (Highlighted information when [...]

Aw that's sad :(. unfortunately I am not connected to a university. Any license sharing possibilities out there? The research I am doing on the topic sure validates a license... I dont understand why they are not more responsive at matlab.

frederiknolte, 2023-05-04:
I actually called them on the phone once and they were much more responsive!

frederiknolte, 2023-05-04:
Yes